winder/Universal-G-Code-Sender #521: UGS platform #670 update

Encarnacao, 2016-10-28:
[image]

where is the buttons to conect machine and port choise?

winder, 2016-10-28:
Oops sorry about that. The little red circle in the bottom right has the error message, but I was able to reproduce the issue as well.

The worst part is that the feature I was adding didn't even work that well:
[image]

Also new buttons you can add keyboard shortcuts to:
[image]

Encarnacao, 2016-10-28:
thanks for the fast update, if you dont forget if would be possible to be possible to add the other widget panel to that widget on top like for exemple"files browser"
that would rox

closed 2016-10-28 by winder