winder/Universal-G-Code-Sender #522: common actions widget

Encarnacao, 2016-10-28:
the new versions of the UGS platforms now have when you connect machine you press for exemple $X comand on common actions widget to unlock and then you cant acess that widget any more unless i go to jog widget and make machine do some movement and only then you can acess common actions widget, this can be bery annoying, didnt use to be like this, is this a bug?

winder, 2016-10-28:
Thanks, I'll take a look. There are a couple issues related to this I've been thinking about. I used to track the state by saying `If you press the send button the machine is sending`, but have been moving towards using `grbl state = running, so the machine is sending`. There are some weird cases when these two modes interact that I need to fix.

winder, 2016-11-01:
@vMeph I'm almost finished with this, hoping to get it finished in the next day or two.

vMeph, 2016-11-01:
awsome great news mate:)

closed 2016-11-01 by winder

vMeph, 2016-11-01:
just tested the commit you did and still same you press $X and you blocked untill you go jog something
sorry.
Winder sorry for the question but in commits you have a dot green but somethimes is red, red means failure but in reality what realy fails?does that mean not a good version to download?

winder, 2016-11-01:
@vMeph you're right, I didn't notice the red x above until right after pushing the change. I just fixed the test that was failing, you should be able to download the updated build now.

vMeph, 2016-11-01:
@winder ok just have tested really all seems good now.
just to let you know spindle,flood and mist buttons are in red squares not sure if is supose to be like that.
thanks alot for the fast fix, awsome job mate 👍

vMeph, 2016-11-01:
@winder 1 more question do you think that if would be a good idea and if possible that jog controlls keep the last settings you have chose like the mm steps and the feed rate?
that way there wouldnd be no need to change all the time.
dont know if feed or steps makes sense but for sure the mm or inch would make sense, i personaly always use mm and have to go there change all time sense it starts with inch  hehe

commit 4e1da335 (2016-11-01): GrblController reports state changes when status message indicates a
change has occurred. Fixes issues when clearing alarm, and allows
pausing during a jog. Closes #522