winder/Universal-G-Code-Sender #524: Serial UnsatisfiedLinkError(Missing Dependencies ???)

jobongo, 2016-11-01:
I'm trying to install the UGS Platform on a arm based board(Pine 64 running Debian Mate) and keep running into and issue with jssc. I have installed the platform on RPi 3 and it works (somewhat, the visualizer doesn't has issues). On the Pine64 I get the following error

`java.lang.UnsatisfiedLinkError: jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J
	at jssc.SerialNativeInterface.openPort(Native Method)
	at jssc.SerialPortList.getUnixBasedPortNames(SerialPortList.java:333)
	at jssc.SerialPortList.getPortNames(SerialPortList.java:296)
	at jssc.SerialPortList.getPortNames(SerialPortList.java:178)
	at com.willwinder.universalgcodesender.utils.CommUtils.getSerialPortList(CommUtils.java:45)
	at com.willwinder.ugs.nbp.core.connection.PortComboBox.initComboBox(PortComboBox.java:53)
	at com.willwinder.ugs.nbp.core.connection.ComboWithPreferences.<init>(ComboWithPreferences.java:87)
	at com.willwinder.ugs.nbp.core.connection.PortComboBox.<init>(PortComboBox.java:45)
	at com.willwinder.ugs.nbp.core.toolbars.SerialPortToolbarBox$MyToolbarPresenter.initComponents(SerialPortToolbarBox.java:103)
	at com.willwinder.ugs.nbp.core.toolbars.SerialPortToolbarBox$MyToolbarPresenter.<init>(SerialPortToolbarBox.java:83)
	at com.willwinder.ugs.nbp.core.toolbars.SerialPortToolbarBox.getToolbarPresenter(SerialPortToolbarBox.java:66)
	at org.openide.awt.Toolbar$Folder.createInstance(Toolbar.java:443)
	at org.openide.loaders.FolderInstance.defaultProcessObjectsFinal(FolderInstance.java:888)
	at org.openide.loaders.FolderInstance$1R.run(FolderInstance.java:730)
	at org.openide.util.Task.run(Task.java:257)
[catch] at org.netbeans.modules.openide.loaders.AWTTask.run(AWTTask.java:79)
	at org.netbeans.modules.openide.loaders.AWTTask$Processor.run(AWTTask.java:172)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
` 

I have the feeling that there are some missing dependencies in the distro that I am using on the Pine64. UGS opens but the serial port selection as well as the visualizer are not visible. 

UGSPlatform works fine on desktop Ubunutu and Ubuntumate for RPi, but i can't seem to get it to work with Ubuntu Mate on the Pine64. I am using openjdk 1.8. I have the rights on the serial port. There is something I am missing. 

BTW works great on Windows and Ubuntu GRBL CNC Machine. I am trying to create a "standalone" CNC and this software seems to be the best solution. Thanks.

winder, 2016-11-01:
It sounds like the libraries I'm using for Serial (jssc) and OpenGL (jogl)
don't support your hardware, at least not in the way I'm using them.

Does the arduino IDE (which also uses jssc) connect to your arduino?

jobongo, 2016-11-01:
Arduino connects and I can send GRBL commands just fine. It seems that UGS creates a lock file for the TTYACM0 because I get the error that a lock file already exists....

winder, 2016-11-01:
What version of UGS are you using? If you aren't already, could you try the
nightly build?

jobongo, 2016-11-01:
I am using version 2.0. To be honest I haven't tried the previous versions (0.9, 0.8, etc). I have tried the nightly builds of v2.0 and the stable version (I assume it is the stable version). Both give me the same error. I like the look and design of v2.0. I will try the latest nightly that was just posted.

jobongo, 2016-11-03:
I think I have found a possible cause. The Pine64 CPU architecture is arm64 while the RPi is ARM-HF. In the jssc SerialNativeInterface class, it doesn't check for arm64 architecture, only armhf and armsf (soft-float and hard-float) and  or x86/64 then loads the correct shared object (.so) (I know very little Java, but from what I can see, this is what happens). 

I found a version of JSSC that support arm64 but since UGS is not compiled with this is there any way to replace the Java classes with the ones from jssc arm64? I have searched for the jssc classes that are called but can't seem to find the correct ones. The only place I could find them is in the jssc.jar that is located is 

`ugsplatform/modules/ext/com.willwinder.ugs-platform-ugslib/org-scream3r/`

but changing these classes seem to have not affect. I have even deleted this jar and get the same error. Could these classes have been placed in different jar or somewhere else? Thanks.

winder, 2016-11-03:
Where did you find a version of JSSC for arm64? If there's a build with wider platform support people are using I should switch to it.

The problem with these libraries is that the java environment doesn't have built in libraries, so they need to bundle in some native code. So you need to replace the class files AND add in the native binaries. If you open up the jar (they are zip files), you can add in the library files and update the class files:
[image]

jobongo, 2016-11-03:
I found the deb package for jssc arm64 at [https://launchpad.net/ubuntu/+source/jssc/2.8.0-1/+build/9880902](url). 

I have tried updating the class files and the libraries like you mentioned, but I get the same error. Even when I delete the jssc.jar, I get the same error. This makes me think that I am looking in the wrong place.

jobongo, 2016-11-03:
I found that it does use that jar. When UGS opens, it creates the .ugs and .ugsplatform directories in the home directory. There must be some cache file there that defines the previous state. When I clear those directories, I get an undefinedclasserror, different from the one above. 

I can't find the source for the deb file that includes the arm64 architecture. I gonna see if I can modify the SerialNativeInterface class to include this architecture and include the correct libraries.

jobongo, 2016-11-06:
I managed to resolve the issue. I used the shared object from the deb package (libjssc-2.8.so) before and copied it in the `/usr/lib/jni/` directory. Then I just replaced the original `SerialNativeInterface` class located in `ugsplatform/modules/ext/com.willwinder.ugs-platform-ugslib/org-scream3r/` with the one in the deb package. Cleared the ugs* hidden files in the home directory and I am now able to communicate with my CNC. 

I think that modifying the SerialNativeInterface class would not be hard to include arm64/aarch64. I started to but really just wanted to find and easier way.... In the deb arm64 repo, version 2.6 is used, not v 2.8 so you would have to include this in your jssc.jar.

closed 2016-11-06 by jobongo

fbmnds, 2020-11-02:
This also worked on a Jetson TX1.

J-Cat, 2021-01-04:
This also worked for me running Ubuntu 20.04.1LTS on Raspberry Pi ARM64.

I followed the following process:
- install openjdk14 (sudo apt-get install openjdk-14-jdk)
- install lbjssc-java (sudo apt-get install libjssc-java)
- installed ugsplatform to /opt/ugsplatform
- copy /usr/share/java/jssc.jar to ugsplatform/modules/ext/com.willwinder.ugs-platform-ugslib/org-scream3r (cp /usr/share/java/jssc.jar /opt/ugsplatform/ugsplatform/modules/ext/com.willwinder.ugs-platform-ugslib/org-scream3r)