winder/Universal-G-Code-Sender #526: UGS inicialize

vMeph, 2016-11-04:
@winder when inicializing UGS platform you get folowing output from grbl settings, just wondering why in midle of settings after $25 shows line >>> $G any particular reason or just normal?
`Grbl 1.1d ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 2    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 500.000    (Homing locate feed rate, mm/min)
$25 = 2000.000    (Homing search seek rate, mm/min)
>>> $G   <------------------------------------------------------this area is it normal?
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 160.000    (X-axis travel resolution, step/mm)
$101 = 160.000    (Y-axis travel resolution, step/mm)
$102 = 80.000    (Z-axis travel resolution, step/mm)
$110 = 2000.000    (X-axis maximum rate, mm/min)
$111 = 2000.000    (Y-axis maximum rate, mm/min)
$112 = 2000.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 432.000    (X-axis maximum travel, millimeters)
$131 = 338.000    (Y-axis maximum travel, millimeters)
$132 = 35.000    (Z-axis maximum travel, millimeters)
ok
ok`

winder, 2016-11-04:
When UGS starts up it sends 3 commands to GRBL used to initialize GUI behaviors: `RESET (CTRL-X)`, `$$`, and `$G`.

In this case GRBL has already started responding to the `$$` before UGS managed to send out `$G` so you see the responses mixed up a little. Nothing to worry about!

closed 2016-11-04 by winder

vMeph, 2016-11-04:
@winder couldnt initialize GUI behaviors order be  RESET (CTRL-X), $G. and $$ ?

apreciate the explanation :)
thanks