winder/Universal-G-Code-Sender #527: 2.0 bug, return home from canceled job needs to lift z-axis before moving!

supagu, 2016-11-11:
just noticed this nasty one.
if you are doing a cut, and you cancel it. Then hit return home it doesnt lift the zaxis up until it gets into the home position. If the machine is not turned on this could break the bit!

vMeph, 2016-11-11:
i just have test and all works fine if doing a job and you hit stop and then send home ($H) z axix move first and then goes x and y as supose, unless you have changed config.h 
#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
thats the original  will move first Z and then x and Y

vMeph, 2016-11-11:
ups my bad i dont use the 2.0 version i use the UGS Platform, so not sure how performs in version 2.0
https://winder.ci.cloudbees.com/job/UGS/
witch in my opinion is a better controller
[image]

winder, 2016-11-17:
@supagu if UGS see's the machine is at `Z < 0` it will raise the tool to zero before going to (0,0,0). Hope that helps!

winder, 2016-11-17:
@vMeph this issue was with the `Common Actions` box in your screenshot, the `Return To Zero` button in there.

commit fe0e040e (2016-11-17): Return to zero will raise tool to Z=0 if it is less than zero. Closes #527
closed 2016-11-17 by winder