winder/Universal-G-Code-Sender #530: Laser stays on with pause

DOUG888, 2016-11-12:
Hi Guys
Running a laser program and the power board apparently was overheating (first time) pressed PAUSE to give it a chance to cool down but it stayed running at S1000...need an option to set S..... value to Zero while in PAUSE same would apply to a milling operation, for safety if no other reason.
Thanks
Doug

jahnj0584, 2016-11-12:
Maybe for laser, but when milling you must leave the spindle on as there is acceleration time to restart the spindle on a resume

DOUG888, 2016-11-12:
Yes…makes sense…mostly laser for me…so still request that this be added, thanks

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

winder, 2016-11-12:
Laser mode in GRBL v1.1 might do this already.

DOUG888, 2016-11-12:
OK…How do I activate “”Laser Mode”” or are there specific commands that are to be used???

Or does config.h need to be specific for Laser and then another config.h for mill??, Hope this is not the case.

Thanks

 

https://www.avast.com/antivirus

winder, 2016-11-12:
It is the $32 setting. Details here:
https://github.com/gnea/grbl/blob/edge/doc/markdown/settings.md

I don't see anything specific about what happens during a pause. If it doesn't disable the laser that would be a good grbl feature request.

DOUG888, 2016-11-12:
Thanks Will

Yep, can see the $32, but doesn’t specify the PAUSE question…have asked the question on the GRBL wiki, get back to you when I get an answer.

Thanks

Doug

 

https://github.com/gnea/grbl/blob/edge/doc/markdown/settings.md

https://www.avast.com/antivirus

closed 2016-11-13 by DOUG888