winder/Universal-G-Code-Sender #531: Suggestions for joging control

jglazer63, 2016-11-13:
I have used UGS pretty extensively.  Lately, the 2.0 new interface.  I would like to offer a few suggestions that I think would improve the jogging interface substantially.

First off, I want to say I love the software.  I use it pretty much exclusively with my X-Carve.  I purchased a BLUETOOTH NUMERIC KEYPAD that I have attached to the table that the machine is on.  This makes jogging and homing the machine much easier but could still be improved.   Let me lay out my ideas:

1. Double-tap #5 takes the machine to the home position.  Not being able to get to home from the keypad is painful.  Double-tap ensures that you really mean to go home (eliminating the accidental home)

2. While on the subject of home.  I cannot tell you how often I accidentally hit home while the bit was below zero thus ramming the bit into the side of the piece being made and throwing off the machine calibration.   Isn't there any way to have home bring the bit above zero before moving x or y?  For safety?

3. Jog feed increments can be adjusted with "+" and "-".  I am thinking in mm.  So if you're at 1mm and you hit + it goes to 2mm then to 3mm and so on.   Should not need to go any higher increments.  This, of course, could be adjusted in preferences.   Pressing "*" sets the increments to 1mm.  Hitting "-" from 1mm reduces to .1mm then again to .01mm up to .001mm or something.

I would love to figure a way to press the arrow keys or the 9/3 to make the machine jog continuously until you stop pressing them.  This is already nearly the case in a 1mm increment and with keyboard repeats but UGS still stumbles now and then.

Anyhow, being able to use my bluetooth numeric keypad more exclusively would be a wonderful improvement.

Hope these are ideas find popularity!

jon

winder, 2016-11-13:
Thanks for the suggestions. Some are already possible with the platform
GUI. Inside the preferences is a key mapping section so you can make
bindings for increase/decrease step size. You can also keybind a macro that
raises the tool before going home.

GRBL v1.1 has some ways to simulate continuous jogging that I'd like to
explore eventually.

jahnj0584, 2016-11-13:
Some of these can already be done. Homing order of axis should be an option
by now, for sure. Z(machine 0), then XY, then drop the Z like industrial
machines.

Haas' have a jog lock key. Select direction and lock it. It moves until you
say stop. I assume we could do that by sending the same code as the pause
button does now with clearing the buffer?

jglazer63, 2016-11-13:
Home: Z 0 +safety then XY

jahnj0584, 2016-11-13:
Machine 0 will be the max height it can be anyways

vMeph, 2016-11-13:
@jglazer63 intersting idea in using BLUETOOTH NUMERIC KEYPAD , could you give any information how it works and what wiring is need to make it happen and what kypad you using ?

jglazer63, 2016-11-14:
Its pretty simple.   Nothing more than a second keyboard on your computer.   Do a search for you and you'll see what it looks like.   Mine is mounted on a post attached to the table where my x-carve is located.  Lets me jog the machine without using a big keyboard.

winder, 2017-10-13:
superseded by #779

closed 2017-10-13 by winder