winder/Universal-G-Code-Sender #532: Minor interface adjusments

jglazer63, 2016-11-13:
The new interface is looking good but I have 2 complaints and one issue:

1. The "Reset Zero" button is too close to the "Return to Zero" button and I've hit one or the other by mistake which pretty much hoses the piece I am working on.   In my opinion, the "Reset Zero" button needs to be far away from everything else so you know you really mean to hit that.  Maybe even with a dialog that says "Are you sure?"

2.  The "SEND" button should not be in the same vicinity as all the other file operations.  I hit send once by mistake as I was tying to press browse.   What a mess!!!

3.  My monitor, attached to the wall of my workbench, is a small(ish) TV that has a resolution of 1360x760.   The current interface does not stretch well to this size.   It would be nice if the interface was a bit more tolerant of odd sized or smaller monitors.  There is a lot of wasted space in the machine control area that could be condensed.   The check boxes for "scroll output window" and the others could be moved to an options window since I doubt many people change these and its a waste of real estate for them to be on the main page.  By the way, you cannot close "sender settings" dialog with the "X".

Thanks for the good work!!

jahnj0584, 2016-11-13:
I agree and have suggested similar. Reset should be beside the machine
coords beside each axis

breiler, 2020-12-09:
These are solved in the platform edition, hope you think it's ok for me to close. 
The toolbar and new toolbox are editable, so you may now decide which buttons that should be visible and where.

closed 2020-12-09 by breiler