winder/Universal-G-Code-Sender #533: Cannot find java 1.7 or higher.

jglazer63, 2016-11-13:
Whenever I try ugsplatform64, I get the above error.  Nothing I try can point it to where java is installed.  If I open a command window (windows) and type Java -version it shows 1.8.   What am I missing?

winder, 2016-11-15:
@jglazer63 my guess is that you might have the 32-bit version of java installed, when you run `java -version` does have `64-Bit` at the end like this? Does `ugsplatform.exe` run?
```
>>> java -version
java version "1.8.0_60"
Java(TM) SE Runtime Environment (build 1.8.0_60-b27)
Java HotSpot(TM) 64-Bit Server VM (build 25.60-b23, mixed mode)
```

closed 2016-12-15 by winder

gunnerap, 2018-10-27:
i am having the same issue but i do have a 64 bit version of java. i am brand new to cnc. this stinks prety bad. my machine is almost ready to use and i cant open platform. any help is greatly appreciated

multishape, 2019-05-21:
I'm getting the same error. I've tried to add a new java locations, and also edit the location of JDK on the script. Still having the same error.

breiler, 2019-05-22:
@multishape What is the output if you run `java -version` from the command line?
Are you trying the latest nightly build of UGS?
Which version of the operating system are you using?

Petesystemspecialist, 2020-09-08:
Now in September 2020 latest UGS platform , windows 10 64 bit and still getting the same thing after installing Java 64 bit

breiler, 2020-09-08:
@Petesystemspecialist are you running the latest version from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

There is a windows package which includes Java so there is no need to download it separately (that is why this issue is closed).

Petesystemspecialist, 2020-09-08:
Hi there

And many thanks for your reply.  I kept missing this one. Being going nuts
for days
This platform version is so much better!
Thanks once again

Peter Ford