winder/Universal-G-Code-Sender #538: Optional seperate Z axis

jahnj0584, 2016-11-20:
well, i might be the only person who doesnt like the new optional jog Z settings! Can you PLEASE make this an optional thing? I broke a bit tonight because while using 0.1mm Z I expected XY to do the same but they were still on 10. I'd much rather the space be used for 0.1/1/10 selectable increments or not at all.

aforww, 2016-11-24:
I absolutely agree with this. I use an xbox controller mapped to keyboard shortcuts to manage everything and now I canI because I have no way of mapping step adjustments to z. Please make this an optional change.

closed 2016-12-01 by winder

aforww, 2016-12-03:
I see you closed the issue.  however, I'm new to this and don't know what to make of the changes.  I know I still can't adjust all three axis simultaneously.  Is there a setting for this now?

jahnj0584, 2016-12-03:
it seems like you have an option in the settings window. You have to re-download the file or read #546

aforww, 2016-12-03:
Thank you. I will try and figure it out.  Again,  new to this.  What do you mean download the file?  The nightly builds or something else?

jahnj0584, 2016-12-03:
yep, just off the front page here in github! The zip for ugsplatform.

winder, 2016-12-04:
@aforww this doesn't let you jog in multiple directions at the same time, it lets you use a single step size for XY and Z.

[image]

aforww, 2016-12-04:
By "adjust"  I meant change step size for all three.  But I found the settings about twenty minutes ago.  Thanks for the reply though,  I really appreciate it!

commit dd62cb88 (2016-12-01): Separate XY and Z step sizes is now a configurable setting. Closes #538