winder/Universal-G-Code-Sender #540: Move reset X/Y/Z buttons beside work 0

jahnj0584, 2016-11-20:
I think it would make much more sense for them to be there, and not in the common actions window. Machine coords should also just be a subscript underneath as it's only used for reference if using jigs + homing. 

Ideally, it'd be great if EVERY button could be moved around as docked as we see fit, then locked into our own "windows" that we manage (ie jog controller, coords, file mananger, etc)

closed 2016-11-20 by winder

winder, 2016-11-20:
Here's the new panel. I added the pin states tonight as well, the panel is getting pretty big though so I might make separate windows for the DRO and the state/comment/pin.

[image]

jahnj0584, 2016-11-20:
What are the pin States for?

jahnj0584, 2016-11-20:
But yes seperate would be good. I feel like I won't be using them

winder, 2016-11-20:
They tell you when the various external switches GRBL supports are activated. Like the probe, or XYZ endstops.

jahnj0584, 2016-11-20:
Does it just go by what's plugged in? I just ordered an E-stop but if it
would be programmed to just pause the program and spindle that'd be gravy

winder, 2016-11-20:
Yes, it's just for reporting. Sonny is adding a feature to pause the spindle (laser) when laser mode is enabled. Maybe you could ask him to bring that feature to non-laser mode.

commit c106f1b9 (2016-11-20): Move reset XYZ buttons to the DRO panel. Closes #540