winder/Universal-G-Code-Sender #541: Rapids speed overrides

jahnj0584, 2016-11-20:
I think this should be done on a percentage based control, instead of 10/50/100% values as currently configured.

winder, 2016-11-20:
I was hoping all 3 modes could be percentages like that, but the implementation of GRBL would make it pretty difficult to do differently than how I've got it now.

closed 2016-11-20 by winder

jahnj0584, 2016-11-20:
would it be easier to atleast make them configurable? I'd rather something like 40/75/100 or something then the current.

winder, 2016-11-20:
It's a GRBL setting, you can change it but the GUI will still say 25/50/100. In `config.h`:
```
#define DEFAULT_RAPID_OVERRIDE  100 // 100%. Don't change this value.
#define RAPID_OVERRIDE_MEDIUM    50 // Percent of rapid (1-99). Usually 50%.
#define RAPID_OVERRIDE_LOW       25 // Percent of rapid (1-99). Usually 25%.
```

jahnj0584, 2016-11-20:
Wicked! works for me. I rarely touch the rapids but its good to know.