winder/Universal-G-Code-Sender #542: Return to 0 axis order

jahnj0584, 2016-11-20:
Still having issues where it moves Z to work 0 first. Please just revert this to be XY then Z, OR raise it by a user configurable amount.

winder, 2016-11-20:
Z should only go to zero first if it was < 0 before, otherwise it should behave the way you described.

jahnj0584, 2016-11-20:
it didnt today, when I just re-downloaded it. I made sure to connect with the Z at the top to set a high Z0 too!

DirkHeinke, 2016-11-22:
Probably the problem is in in GrblController: https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/GrblController.java#L354

`this.machineLocation.z` is used, but in my opinion this should be `this.workLocation.z`?

aforww, 2016-11-24:
I am having the same issue. It is not existent on my other machine running a different version. I have homing switches installed and home every time I turn the machine on. So, Home, jog to desired work coordinates, reset zero. From there, anytime I hit "return to zero" it drops Z-axis first onto the work piece then drags it to X and Y zero. The G90 G0 Z0 command being issues before G90 G0 X0 Y0 Z0 needs to be changed.

DirkHeinke, 2016-11-24:
Do you see two G90 G0 Z0 commands?
```
G90 G0 Z0
G90 G0 X0 Y0
G90 G0 Z0
```
That is what I see at my maschine.

aforww, 2016-11-24:
No,  just the one. I thought the one was off,  but having two.  That makes
no sense.

winder, 2016-11-28:
@DirkHeinke I just noticed your comment from 6 days ago. I think you're right and I've updated the code accordingly. Let me know if this helps the situation.

DirkHeinke, 2016-11-30:
Yes, I just downloaded the latest nightly build and tested it. First X/Y-axis, followed by Z. Perfect, thanks for the fix.

closed 2016-11-30 by winder