winder/Universal-G-Code-Sender #544: new jog issues

jahnj0584, 2016-11-20:
since the new jogs, you can no longer send a continuous stream. It sends, locks, then unlocks the controller when the first command is finished sending. This is REALLY annoying and makes for a very cumbersome and jerky style of jogging. The old way was A LOT better, and if it means removing the feed command that has been added, then I accept the small sacrifice.

winder, 2016-11-20:
I made some tweaks a couple nights ago to fix the weird GUI state changes, you should be able to queue up a bunch of jogs in a row now.

jahnj0584, 2016-11-20:
I was using a fresh download from today

vMeph, 2016-11-21:
i have new version and also same problem you cant send a continous move on jog

winder, 2016-12-15:
@jahnj0584 @vMeph This should finally be fixed.

commit 2e89ebd0 (2016-12-15): Fix possible jerky jogging. Closes #544
closed 2016-12-15 by winder