winder/Universal-G-Code-Sender #545: UGS platform Feedrate

vMeph, 2016-11-20:
when you first connect machine the feedrate is set to 10 on jog controler and if you send command lets assume F1000  nothing changes, same  if i do a macro button with F1000 when machine first connect you got no acess to macro butttons, nothing changes unless you go to jog controller and change the feed rate, it didnt use to be like that, so everytime i start machine if i want to me arround im force to go to jog control to make changes when i could have a easy acess by making a macro button with F1000 and just click and star moving arround or just type a F1000 on command line

it makes sense that it should let you acess macro buttons when you first connect machine?
it makes sense that it should let you change feed rate by using a macro button or by  sending a command?

jahnj0584, 2016-11-20:
Did you send $X first?

vMeph, 2016-11-20:
i did do you have same problem or works fine?

jahnj0584, 2016-11-20:
I didn't have this issue yesterday but I do think the feed rate option for
jogs should be removed or only set as 100/75/50 of your X/Y/Z rapid speed
set by the $ configuration

winder, 2016-11-20:
@vMeph I found the feed rate issue, it wasn't being restored or updated properly. I don't follow the issue with macros though. They don't enable for you when the machine starts?

What version of GRBL are you using?

vMeph, 2016-11-21:
yes what i mean is when i start machine they are enable, but sense i have switches i got to first remove alarm $X and then macros buttons become disable, they only be enable if i do a jog move after

winder, 2016-12-15:
@vMeph this should be working, let me know if you're still seeing problems.

closed 2016-12-15 by winder