winder/Universal-G-Code-Sender #549: VISUALIZE

RWW549, 2016-11-27:
I've used UGS without problems until I upgraded to Windows 10.  The visualizer always had thin precise lines that were easy to keep track of, but since the upgrade, all the lines in the visualizer are many pixels wide and not much better than a blur. the space between the lines is smaller than the lines are supposed to be. No other programs were effected, and I can't find any fix on the windows side. I downloaded a new copy of UGS with the same result. Any suggestions?
[image]

jahnj0584, 2016-11-27:
its time for you to use the platform anyways.

winder, 2016-11-27:
@rww549 I'm not sure what causes this, but it's been a problem pretty much
since day 1 of the visualizer for some people on windows. The ugs platform
visualizer has been almost completely rewritten and seems to be more
consistent across different operating systems.

RWW549, 2016-11-28:
Thanks for the quick reply. It's still worth a lot more than I paid for it, but if you happen to find a fix, let me know.    Thanks.

closed 2016-11-28 by RWW549