winder/Universal-G-Code-Sender #551: Random lockups - check operating system power saving settings

splitn2, 2016-12-01:
Hi ,

I have a question and perhaps an answer for those who have had occasional lockups with UGS or any other gcode sender.

I have helped a few friends with their small machines get started with GRBL. I keep them up to date with the latest Platform UGS and GRBL in sync with the recent changes to ensure its as generally up to date and bug free and supportable as possible. However I was getting friends frustrated with lockups that apparently happen the moment I turn my back!!!! 

Wiring issues aside and I guess the possibility of a bad Arduino Uno or bad usb port, there seemed to be a pattern to this and the result is that we appear to have much improved stability by changing the Windows power plan settings to "maximum" so that the cpu never sleeps or throttles back.

Does anyone else have similar experience ?

I thought this was worth sharing in case it saves someone a lot of time and effort chasing an issue that is not actually a fault.

cheers
Richard

winder, 2016-12-02:
The latest version has a "keep alive" mode when you're sending a file. Basically it moves the mouse one pixel to make sure the computer doesn't fall asleep. I'm not sure what could be locking up though, does it happen during a job?

splitn2, 2016-12-02:
yes in the middle of a job although I have never seen it in person just had frustrated users messaging me of their woes! For one guy it was generally after a period of 20 mins , and he';s not so computer savvy so couldn't put it in a technical way he more explained "it jammed up so I rebooted everything"  but after some digging we decided to do the usual things like pause/stop anti virus and put it on maximum power mode where the cpu doesn;t throttle back, the sort ofthings I would do on MACH3 pc for example if I want good real time cnc behavior (as good as one would expect from windows anyway) . The keep alive is interesting and probably does help I'll make sure we consider that too in future. cheers

closed 2017-05-06 by winder