winder/Universal-G-Code-Sender #553: Cannot jog less than 1mm

jglazer63, 2016-12-04:
I appreciate fixing the keyboard jog but now I cannot go less than 1mm. Even though it says .001mm it still seems to go much more than that.  Nothing changes from 1-0mm.

ganeby, 2016-12-14:
Hello

has no-one answered you yet?

Grant

winder, 2016-12-14:
I haven't had time to look at this one yet

jahnj0584, 2016-12-14:
It has to be his side or from the latest version I always use 0.1mm when
setting my zero

ganeby, 2016-12-14:
Hello, I think it is grbl.

chamnit, 2016-12-15:
The only reason Grbl would not move less than 1mm is if the steps/mm setting is >= 1. There isn't much else that would cause that type of behavior.

winder, 2016-12-15:
@jglazer63 I can't reproduce this in the Classic or Platform versions. If you are typing in a number you might have to tab out of the text box in order for it to register. Let me know if you're still seeing issues.

closed 2016-12-15 by winder