winder/Universal-G-Code-Sender #555: Feature request: sending email/text when job is done

shayo, 2016-12-16:
Would be nice to have a way to send an SMS or an email when jobs are done so I don't have to come and check the machine :)

jahnj0584, 2016-12-16:
Can you not just enable sleep mode for GRBl and set a timer?

winder, 2016-12-16:
It's an interesting suggestion, but I'm not sure I want to encourage people to leave their machines unattended.

shayo, 2016-12-16:
I'm doing aluminum with Xcarve so I need to go very slow.... jobs can take
couple of hours. I'm not going to babysit my machine :)

Shay Ohayon
HHMI Postdoc Fellow of the Life Sciences Research Foundation
DiCarlo's lab, MIT, 46-6147
+1 617-324-3593

aforww, 2016-12-16:
Invest in a web cam then.  Problem solved.

shayo, 2016-12-16:
Workaround... My Form2 3D printer has no problem sending me emails when
jobs are done.

Shay Ohayon
HHMI Postdoc Fellow of the Life Sciences Research Foundation
DiCarlo's lab, MIT, 46-6147
+1 617-324-3593

closed 2016-12-17 by winder