winder/Universal-G-Code-Sender #556: Feature request: Reset Z axis to another number than zero

poisa, 2016-12-23:
As shown in [this YouTube video](https://www.youtube.com/watch?v=Q-Q1o16Ux-I) it is sometimes useful to reset your Z axis to some arbitrary dimension other than zero in order to prevent cutting through your spoilboard. So instead of telling your machine "hey, this is zero" while the bit touches the top of your work, you tell it "hey, this is 1 inch" when the bit is touching the spoilboard. This makes it very easy to ensure that your cutter will not go into your spoilboard.

I'd love to be able to do this with Universal G Code Sender.

poisa, 2016-12-23:
>or simply jogging that 1" off your spoiler and using the existing button to reset it instead of the material.

Sometimes the easiest solutions are right in front of us. Thanks!

closed 2016-12-23 by poisa

winder, 2016-12-24:
@poisa thanks for the suggestion. I think right now I'd recommend a macro, I think what you want is something like `G10 P0 L20 Z1`. You can also try `G10 P0 L20 Z{prompt|z-height}` if you find yourself wanting to use different heights.