winder/Universal-G-Code-Sender #558: Baudrate 230400 needed

tymop, 2016-12-26:
The baudrate for arduino grbl is 230400 baud. I'm able to connect and send Gcode by hand on the serial port. But I can't connect with Universal-G-Code-Sender because the maximum baudrate is 115200.
I do not know how to add it.

Please add 230400 baud and maybe the others higher.

thank you!

winder, 2016-12-27:
@tymop Added 230400. FYI that field is editable so you can workaround the issue if you are testing something with an even higher baudrate.

closed 2016-12-27 by winder

tymop, 2016-12-28:
I can not see the new baudrate.
I downloaded the zip file, and replaced my old folder, but I can not see the 230400 baudrate.

You said I can edit it. But how ? It is impossible to modify the field.

winder, 2016-12-29:
@tymop You can edit the field in the "platform" version. The nightly builds have this change in it.

TamGit, 2017-02-07:
Good morning,
I'm also interested in modifying the baudrate at higher value than 230400 but I can't find where to edit it ?
It's about laser engraving where the segments are way smaller than when milling stuffs.
Thanks.

winder, 2017-02-07:
@TamGit This option is in the nightly build version at the bottom of the page here: http://winder.github.io/ugs_website/download/

TamGit, 2017-02-07:
Ok I got it. I thought it was about a file to modify but it's the baudrate field that is directly editable.
Perfect !
Thanks.

awd86, 2017-07-24:
So I'm having a similar issue just trying to set 115200. Sorry to resurrect the thread, but can anyone help me find where to edit the baud rate in the files? 

I can't use the mouse to select any options below the boundary of the window. Also, i can't edit the baud rate field. I'm using Ubuntu on an old Optiplex 330

vMeph, 2017-07-24:
@awd86  
GRBL by defaul comes with  115200 baudrate already
so most like you have to change in the UGS the baudrate to 115200 also, i think comes at 9600 baudrate by default

anyway if you want to confirm on GRBL files you can look on GRBL folder in libary, the config.h file 
on lines 40,41,42 you will see this

// Serial baud rate
// #define BAUD_RATE 230400
#define BAUD_RATE 115200

like you see is already set to 115200 baudrate by default

winder, 2017-07-25:
@awd86 it sounds like you might be using an old version of UGS. Try v1.0.9 or the nightly build.

DiegoChajtur, 2020-08-08:
Please add 250000 baud or add editable baudrate field please

breiler, 2020-08-10:
@DiegoChajtur it is now available in the latest nightly build.

commit 1637be5b (2016-12-27): Added baud rate 230400. Closes #558