winder/Universal-G-Code-Sender #561: Step Size Refresh Delay

jahnj0584, 2016-12-27:
so, 99% of the time changing the step size will be "immediate" where if you change it, the next jog command you send will be what you recently entered. BUT, 1% of the time it still "hangs on" to the old step size which results in: broken bits, ruined materials, anger, headaches, etc etc. Is it possible to confirm the change or create some sort of feedback, like a beep on the Haas?

winder, 2016-12-28:
The step size is shown in the status bar at the bottom, but I can look into
a sound effect.

Are you using the keyboard shortcuts when the step size isn't updating?

jahnj0584, 2016-12-28:
With this incident I manually entered the value (0.5 from 20), because
xpadder (Xbox controller) stopped working. I never have issues when I use
the shortcuts via the DPad to adjust to the presets (1/10/100 etc)

closed 2017-10-12 by winder