winder/Universal-G-Code-Sender #563: Unable to conduct soft reset after soft limits hit

Bosque00, 2017-01-03:
Summary:
When soft limits are reached the UGSplatform disables all buttons in "Common Actions" not allowing a user to conduct a soft reset of Grbl. Currently disconnecting and reconnecting serial port to reset Grbl. This looses the current position.

Hardware:

Shapoko 3
Software:
Grbl 0.9g
Java: 1.8.0_51; Java HotSpot(TM) Client VM 25.51-b03 Runtime: Java(TM) SE Runtime Environment 1.8.0_51-b16
ugsplatform.exe 1/1/2017
System:
Windows 7 version 6.1 running on x86; Cp1252; en_US (serverlaunch)
Steps to create error:

Start Grbl
Start UGSplatform
Home system
Use jog controller to move any axis past a soft limit.
Attempt to issue a soft reset command
Workaround - Disconnect the serial port and reconnect it. Position is lost.

jahnj0584, 2017-01-03:
Try only using UGS and the stop button or $X

Bosque00, 2017-01-03:
This works using the UGS and $X.

kardinal7, 2017-02-15:
The same happens to me in the platform. If limit switch is detected I need to disconnect the UGS platform and re-connect

71tr6, 2017-02-26:
> Start Grbl
Start UGSplatform
Home system
Use jog controller to move any axis past a soft limit.
Attempt to issue a soft reset command
Workaround - Disconnect the serial port and reconnect it. Position is lost.

Same thing here. You don't even have to physically move the axis, just give a g code command that exceeds the soft limit.

aforww, 2017-03-18:
Was there ever a resolution to this?

Bosque00, 2017-03-18:
I have not used UGSplatform since finding this issue as the feature works better in UGS. I will have to try the latest version.

winder, 2017-03-20:
I haven't worked on this issue yet. I recently wire a button to my test arduino so at least I can finally simulate some of these homing/limit scenarios.

aforww, 2017-03-22:
@winder it seems to be isolated to platform. Hopefully it gets worked out.

cjbaar, 2017-04-24:
I am not sure if this is the best place to comment on this, but I am having a similar issue. After hitting a hard reset (running into a physical limit switch), I have to disconnect and reconnect the serial port in order to reset the arduino firmware. However, about 25% of time, the "Jog Controller" functions do not re-activate. Even after homing or issuing $X, they are still grayed-out, and I have to quit and re-launch. I am using the platform version on OSX.

winder, 2017-07-09:
Thanks everyone for bearing with me. This issue is finally resolved.

[image]

closed 2017-07-09 by winder

aforww, 2017-07-09:
You sir, are the man.

commit 46681f4c (2017-07-09): Better handling for ALARM states. Closes #563