winder/Universal-G-Code-Sender #564: G02 ERROR

DOUG888, 2017-01-12:
Just running the new setup, latest GRBL and Platform.
Repeating error at the same code location, have attached screen shot.

Ran the program twice, exactly the same spot, same error.

Also cannot see the XYZ symbols in the DRO next to reset buttons. see same screen shot

Apart from that all seems OK..
Thanks for the hard work.
Doug
[image]

jahnj0584, 2017-01-12:
Your column width is probably too narrow. I also had issues sometimes with
the long strings of commands, even they they fit under the settings.

vMeph, 2017-01-12:
Off the topic what ugs skin you using? Where to download? How to install?
Thanks

DOUG888, 2017-01-12:
Jahnj
Thanks for the feedback, but what exactly do you mean by column width??? Which column?

DOUG888, 2017-01-12:
vMeph
This is a theme file called Darcula.jar , I have attached the file but i don't remember how to install it. There are other themes available ...go to  TOOLS/OPTIONS/APPEARANCE/LOOK AND FEEL....YOU WILL SEE THEM THERE.
[com.revivius.nb-darcula.zip](https://github.com/winder/Universal-G-Code-Sender/files/703053/com.revivius.nb-darcula.zip)

winder, 2017-05-06:
I heard recently that this issue happens with fusion 360 when generating gcode in inches. The 33 error means that the arc was not specified accurately enough. If this is still a problem you could try posting it on the grbl github for more details.

closed 2017-05-06 by winder