winder/Universal-G-Code-Sender #565: Universal GCode Sender randomly disappears

MikeMidi, 2017-01-12:
Help - Ever since i bought my milling router CNC machine, the Universal GCode sender randomly disappears leaving my laptop home screen only. When this happens the CNC machine gantry stops. (The spindle remains on if it happens during a toolpath run) I can only gain back control once I restart the Universal GCode sender once again, soft reset and $H. The Universal GCode sender does NOT hang up or 'Freeze', it disappears, shuts down completely.

The power supplying the 3ph inverter that powers the spindle is separated by >50' of cable supplying power to the CNC controller

This happens during a toolpath run, it has also happened when I manually move the gantry using the keyboard arrow keys. When it happens I cannot repeat it. It can happen after a few minutes/seconds of use or after many hours of use, no warning, entirely random. I have changed the controller, and tried a different laptop, it makes no difference. One laptop is AMD powered the other is Intel powered. The spindle can be running or not. This has become incredibly frustrating.

Can anyone offer advice, or have you experienced this at any time before and applied a fix?

Set up:
Java: Ver.8 update 111 build 1.8.0_111-b14
Universal GCode Sender: 1.0.9
Windows 10, Ver.1607 2GBRAM 32bit OS X64 processor based
Processor: Intel Atom CPU Z3735F 1.33GHz

winder, 2017-01-12:
I wonder if it could be some sort of over-zealous anti-virus software? Could you run UGS from the terminal, is there anything printed in the terminal after the program quits?

Alternatively, if you use the UGS Platform version, it should have better error handling for this.

MikeMidi, 2017-01-12:
Hey Winder, thanks for replying... ok so how exactly do I do these 2 things:
 1 How do I run UGS from the CMD prompt.
2 Also there's a UGS Platform version?.. is this something different i need to download and install?

winder, 2017-01-12:
@MikeMidi 
1. If you can get to the directory with the UGS jar run `java -jar UniversalGcodeSender.jar` (or whatever you filename is).

2. Yes, the last link here. It is the new front-end for UGS: http://winder.github.io/ugs_website/download/

winder, 2017-01-12:
@MikeMidi thanks, that's good feedback.  In the zip you downloaded there's a "bin" folder with the executables you need to run in it. Described here: http://winder.github.io/ugs_website/guide/platform/#how-to-run

closed 2017-05-06 by winder