winder/Universal-G-Code-Sender #576: Jog control greyed out

Russellfam, 2017-02-05:
If the Arduino locks up for some reason and I have to unplug it and reconnect, often the jog controls remain greyed out even ifI've reconnected properly to the device. I can home the device and even send files to it but the jog controls remain greyed out.

this happens on UGS and UGS platform

aforww, 2017-03-03:
I'm having this issue every time a job completes or I cancel a job. Did you find a resolution?

winder, 2017-03-05:
@Russellfam I'm not able to reproduce this issue, are you using the latest version of UGS?

Russellfam, 2017-03-05:
I used the nightly builds and since then I've had no issue whatsover

closed 2017-03-06 by winder

jrquinn70, 2021-11-11:
[image]

I have been using gcode sender for 3 years and its worked fine. Today the jog controller is grayed out. Any suggestions?

breiler, 2021-11-12:
@jrquinn70 I have no idea why this happens, but I do recall that we had a problem with the jog controller missing signals that it was connected to the USB controller which has been fixed in later versions.

You could try restarting UGS, then when its started make sure the jog controller is visible before connecting to the machine.

If this doesn't work, close the jog controller window, restart UGS, then open the jog controller from the menu "Window -> Plugins -> Jog controller" and then connect to the machine.

I know that we finally fixed this problem and you could try the latest nightly version. But since you have used this old version of UGS and have not had any other problems with it I am reluctant to advise you to upgrade.