winder/Universal-G-Code-Sender #580: Commands not marked as done, though there is a response

tolland, 2017-02-10:
Hi,
I am using universal gcode sender version 2.0 nightly (5th Feb 2017) with a laseraxe GRBL controller running  Grbl 0.8. I can connect to the controller and send some jog commands, but then it stops and needs to be reset. 

It seems that there is some problem with the format of the response from the controller, as I can see the "ok" response on the console;
[image]
[image]

Here is the console output;
```
**** Connected to /dev/ttyUSB0 @ 115200 baud ****

Grbl 0.8 ['$' for help]
>>> $$
>>> $G
F300S
Grbl 0.8 ['$' for help]
>>> $$
F300S1000{0/0}ok
$0 = 10    (step pulse, usec)
>>> $G
$1 = 255    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 0    (dir port invert mask:00000000)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.020    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 0    (hard limits, bool)
$22 = 0    (homing cycle, bool)
$23 = 0    (homing dir invert mask:00000000)
$24 = 25.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 400.000    (x, step/mm)
$101 = 400.000    (y, step/mm)
$102 = 400.000    (z, step/mm)
$110 = 1500.000    (x max rate, mm/min)
$111 = 1500.000    (y max rate, mm/min)
$112 = 1500.000    (z max rate, mm/min)
$120 = 1000.000    (x accel, mm/sec^2)
$121 = 1000.000    (y accel, mm/sec^2)
$122 = 1000.000    (z accel, mm/sec^2)
$130 = 200.000    (x max travel, mm)
$131 = 200.000    (y max travel, mm)
$132 = 200.000    (z max travel, mm)
{0/0}ok
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F300. S1000.]
{0/0}ok
>>> G21G91G0Z1F10
>>> G90 G21
{1/0}ok
{1/0}ok
>>> G21G91G0Z1F10
>>> G90 G21
{2/1}ok
{2/1}ok
>>> G21G91G0Z1F10
>>> G90 G21
{3/2}ok
{3/2}ok
>>> G21G91G0Z1F10
>>> G90 G21
{4/3}ok
{4/3}ok
>>> G21G91G0Z1F10
>>> G90 G21
{5/4}ok
{5/4}ok
```

tolland, 2017-02-10:
Ah, so this works ok if I use a different controller. So its something about the firmware.

winder, 2017-02-10:
@tolland I'm guessing the command table in UGS is looking for an `ok` response, and the extra output laseraxe added to the `ok` response is confusing it.

Aside from the command table, are you able to send files normally? If so you should be able to ignore the command table.

closed 2017-02-15 by winder