winder/Universal-G-Code-Sender #582: UGS-P - Toggle to ignore certain code?

Durahl, 2017-02-12:
Greetings!

About 2 months ago I started using UGS with an **X-Carve 2.0** but recently switched over to the Nightly UGS-P Build.

My G-Code is being written by Fusion 360 which includes a few lines before the actual project containing a line with G28 I predefined for Tool Changes - G30 for parking the head out of the way.

UGS appears to ignore this first section of code including the G28 Command which I find very welcome as the Machine then moves straight from G30 at the back of the machine, or wherever it is currently parked, to the actual Project code but in UGS-P it also does the Pre-Project code including the G28 Line at the beginning which I find kind of unnecessary as it would then move the machine from the very back to the very front and then back to wherever the work piece is situated.

Is there an option to toggle this **move to G28 at the beginning** behavior off in UGS-P like how it works in UGS?

**Here's a code example:**
```
%
(1001)
(T1  D=6.24 CR=0 - ZMIN=5 - Schaftfraser)
G90 G94
G17
G21
G28 G91 Z0
G90

(1/4 Face)
M9
T1 M6
S16000 M3
G54
M9
G0 X41.355 Y-75.026
etc...
```
I'd rather not remove the lines of Code at the beginning as I fear for them to still serve a purpose :°

jahnj0584, 2017-02-12:
In fusion just disable the "use g28" option when you go to post the code.

aforww, 2017-02-12:
Just to further elaborate on the above suggestion, the G28 option is in the lower right options box when after you click post process in Fusion 360. Additionally, I always change the G54 command of Fusions Gcode to G0 Z10 before saving. That's just a fail safe in case you forget to raise the cutter before traversing.

Durahl, 2017-02-12:
Mhh... Disabling the G28 function entirely also disables it at the end of the Project code which I DID welcome as it would THEN bring it to the front of the CNC where I would then either change the tool or give the machine a small brush over to remove dirt.

I guess it comes either down to removing a piece of code at the beginning, adding some at the end or manually hitting the G28 Macro when needed xD

winder, 2017-02-15:
@Durahl UGS and UGS-P both run with the same processing engine (unless you upgraded from 1.0.9 -> 2.0 at the same time).

You can configure the parser, it sounds like I might have changed one of the default settings. There's a little documentation here:

Durahl, 2017-02-15:
The UGS version I ran before was indeed 1.0.9.

I just took a closer look at configuring the parser but didn't get really far:

- Entering the whole line **G28 G91 Z0** into it would do nothing.
UGS-P would still home towards G28 before heading to the actual project.
I assume this is only meant to be used on a single command like M6 which was already there?

- And then there's the issue with very same **G28 G91 Z0** line also being used to home towards G28 at the end of the project which I still desire to happen so omiting it from the code wouldn't solve the problem.

I guess my hopes concerning this quality of life feature still lie in your hands ;D

Angryfusel, 2017-07-20:
Hey :)
I know the thread is already a few month old but is there any progress in solving that "G28 instead of G28 G91 Z0" problem?
I ran into the same issue with Fusion 360 giving me that line of code at the start / end and UGS Platform just splits it into two lines "G28" & "G91 Z0".
Right now I'm manually editing the code to "G28 Z0" and "G28 X0 Y0".