winder/Universal-G-Code-Sender #584: Z Probing

kardinal7, 2017-02-15:
Is there any way to make a script or a macro to make a Z-Probing?

aforww, 2017-02-15:
I was just researching this last night. Lots of info in the Xcarve and Shapeoko forums . Here's a good one. https://discuss.inventables.com/t/touch-plate/10314/27

But I do wish there was a specific button on UGSP to accommodate it only requiring the user to input bit diameter and plate thickness.

winder, 2017-02-15:
@kardinal7 @aforww Could you guys suggest how this could work in a new UGS platform window? I don't have a lot of time to experiment (and don't have probing on my machine), but I could probably whip something up if you could explain how it should work and what it should look like to be useful.

kardinal7, 2017-02-15:
The Chilipeppr has this functionality working, they call it touch plate
[image]

You need to define the step speed and the height of the touch plate.
After the probe detects the touch plate it corrects the Z axis.

vMeph, 2017-02-15:
Would be great to have auto leveling like chilipepper on UGS platform

jahnj0584, 2017-02-15:
G38.2 then change your dimensions by your probe height.

aforww, 2017-02-16:
@winder I think a simple window allowing the user to input the plate thickness for offset and distance to be traveled before the command cancels due to non-contact would work. You could probably just add a simple button and input pane to the window that already displays the WCS and "reset zero" buttons. 

So it would look something like
Plate Thickness (_Input Value Here_)
Travel to Error (_Input Value Here_)
(Reset Z zero) <---- Button

thibalte, 2017-03-03:
Having looked into it myself I found out you only need to define 2 macros, per @jahnj0584's message:

1. trigger the probing: `G38.2 Z-50 F1` where Z has to be adjusted to probe far enough (but not too long if you have software/hardware limits enabled), and F has to be a reasonable speed for accuracy. Machine will then stop when probe is triggered

2. reset the Z height: `G92 Z39.2` where Z height has to be adjusted for your probe height (mine @is 39.2mm)

voila.

winder, 2017-03-03:
I'm still planning to make the helper window for this but have been very busy recently.

@thibalte thanks for this details on how you use these commands. Would it also be useful to have X/Y probing?

jahnj0584, 2017-03-03:
Of course, but allow for different bit size offsets

winder, 2017-03-03:
I guess for X/Y you would use the bit radius instead of the plate thickness

aforww, 2017-03-03:
Exactly. There's codes to do it via macro. We are just being needy and asking for a built in solution lol.

thibalte, 2017-03-03:
@winder Ah I've never used probing for X/Y, only Z. Actually I wasn't aware that probing worked for X/Y, then it could even be more interesting to support a routine for automatically centering in X/Y with a "corner ring" (not sure about the exact name, basically the bit is jogged in a ring and probes in positive/negative of both X/Y to find the exact center of the ring, the ring having a a corner underside that can be stuck to the corner of the workpiece). Of course that would go on the wishlist, sadly I'm not fluent enough in Java to help on that..

jahnj0584, 2017-03-03:
It's called an edge finder. Its the same probe routing as Z but with X/Y
written in

vMeph, 2017-03-11:
Chilipeppr is open source maybe would be possible to see how they coded to make it the auto level, maybe could eaven make it like they have it and accept files directly from eagle cad to be able to do PCB design, this would be a great feature to have it on UGS and make it a more complete program, other then be jumping software to software to achive the desire things, personaly love it UGS platform just missing some features unfortunatly my programing skills not that good to help out

winder, 2017-03-30:
UGS finally supports probing!  I used this support to build a simple "Probe Helper" which you can find in the latest nightly build menu under `Window > Plugin > Probe Helper`. It seems to be working in my tests, I'll wait for some feedback before closing this issue.

[image]

HomeMaticThomas, 2017-03-30:
This is the Message i got when i was klicking on the "Initial-Probe"
error: Expected command letter: G-code words consist of a letter and a value. Letter was not found.

Do i need to activate the probe feature? 
Where to connect the Probe-Pin?
I am using a Arduino with CNC Shiled V3.0 and GRBL 1.1

thibalte, 2017-03-30:
Probing is active by default. You need to wire the touch plate to GND and the milling bit to pin A5, I use a simple alligator clip with a wire going to A5. (or vice-versa, as long as A5 is shorted to GND when you touch the probed surface).

Regarding the actual error, I couldn't try the feature yet, but I believe UGS will show in the Console what message was sent to Grbl. Can you copy-paste the Console with the command sent to Grbl that is apparently malformed?

HomeMaticThomas, 2017-03-30:
WoW, fast answer:-)!
Thanks for you description to connect the Probe! I have a 3020T and just upgraded it to an Arduino-Version, which ist much better then some other controllers!

This is the complete GCode from the console:

```
> >>> G21 G91 G49
> >>> G38.2 Z-10,000000 F5,000000
> >>> G91 G21
> ok
> error: Expected command letter: G-code words consist of a letter and a value. Letter was not found.
> ok
```

thibalte, 2017-03-30:
oh, I cannot test it but I believe Grbl uses dots for decimal, not comma. Have you explicitly marked `10,000000` and `5,000000` ? if not, then the fields are returning a malformed value. if you manually send `G38.2 Z-10.000000 F5.000000` it should work (at least to start the probing routine, not reset the z-height, you can follow my procedure marked above to probe by sending manually the g-code commands).

HomeMaticThomas, 2017-03-30:
Yes, thats right, the "," is the problem. With a "." it works.

winder, 2017-03-30:
Oops, this is a localization issue. I'll take a look tonight.

rmisko11, 2017-03-30:
Should these be commas?

G38.2 Z-10,000000 F5,000000

HomeMaticThomas, 2017-03-30:
WoW! Great and super-fast work - this is a perfect community!

aforww, 2017-03-30:
Very Nice!

winder, 2017-03-31:
@HomeMaticThomas the comma issue should be fixed now, let me know if it works for you.

HomeMaticThomas, 2017-03-31:
Thanks alot, it works perfect!

vMeph, 2017-04-02:
@winder thanks for have adding Z probing to UGS platform

would be great if UGS could add function of auto leveling also
this is a pefect exemple in what we could do it with auto leveling
https://www.youtube.com/watch?v=ApBfHW7AhsM]

chilipeppr have implemented widged with it
[image]

would this possible to achive?

winder, 2017-04-02:
@vMeph that's the plan.

closed 2017-04-02 by winder

vMeph, 2017-04-04:
@winder from what i see so far probing is working as expected 

now i wonder if this would make any sense
as you know when we do homing the machine will repeat 2 times the process, first will go at fast movement and the will repeat at slow movement to have a better precision
so now lets imagine i did a homing and like you know z will be at the top so start a probing with a feedrate of 10 it will take forever to touch the plate and also when touch the plate it stays down and you will be forced to put z up to remove the plate
so in my point of view would be nice to maybe have lets say 3 input steps on widget , 
1 input for first time that probes to go at faster feedrate, 
2 input to put the second feed rate to go slower rate for better precision
3 input to put value that machine would go up everytime probe toutch plate
so now lets say i set the 3 input 4mm then the sequence would be 1 go faster feed rate touch plate go up 4 mm then go back probe at very slow rate and go up 4 and stop
also if would only filed the 1st input box then would just act has normal like is now, so you would have a choise to do as you prefered

would like to know feedback from what you guys think of this and if it and if makes any sense

jahnj0584, 2017-04-04:
I did this before with a macro. It doesn't change the accuracy of the
probe. Just manually jog 1-6mm above your probe and then use the widget
with one speed. The only addition I'd like to see is a retract height as
well, say up to 30mm despite the probe height being accounted for.

Radiu, 2017-04-06:
Hi i have test the probing,but something is wrong when i do probing toutchs plate and stops but when i start a job file z probe starts going up slowly like its probing and then stars poping errors about m6 then g54 and then all type of errors,basic everything is a error,there was 1 time it worked correctly,the problem isnt the file cause if i run it with out probing works all the time perfectly, any idea what this could be?
Could this be related to noise issues?

cjbaar, 2017-04-23:
This is a great enhancement. Thanks again, @winder. However, I am not 100% clear on how this is used. I have set up a probe, and entered the parameters. When I initiate the probe, it crawls down the Z-axis, as expected, until it contacts the plate. Then, it issues a command like "G43.1 Z-30.925". I do not understand what I am supposed do with that value. As far as I can tell, this only resets the absolute end point of the z-axis, but that does not mean anything to the rest of the gcode. When running a job, it doesn't change anything.

It seems to me, probing should be something like this:
G38.2 Z-10 F1
G92 Z3 (reset zero, based on thickness of touch plate)
G91 Z2 (some kind of back-off so we can remove the plate)

If someone else has a method of using the data from the G43.1 command, please let me know. Thanks!

thibalte, 2017-04-24:
I believe what you just wrote as what UGS should do is what has been discussed above and what it's currently doing, it issues the Probing command (G38.2), and then resets the work Z (G92 Z..). I'm not sure where is this G43.1 coming from?

Also you have a great point about issuing a back-off command after the probing, currently this has to be done manually after the probing, maybe we could add a last argument to the probing panel (back-off distance) and issue that last after resetting Z?

winder, 2017-04-24:
@thibalte @cjbaar I'm using G43.1 instead of G92. That seemed to be the recommended way of doing this.

I agree about the back-off. I'm planning to send the tool back to the starting location rather than having another setting.

vMeph, 2017-04-24:
If we could have the setting for set how mutch  we want to back off tool would be a more complete solution

jahnj0584, 2017-04-24:
I'd also prefer the bit to retract a user-srt distance rather then machine
home or whatever. I probe Z near my work 0 then set x/y

cjbaar, 2017-04-24:
Well, assuming that we are all manually adjusting Z to be "close" to the plate before initiating the probe, the easiest thing to do would be to back off to where the probe started. This doesn't require a setting, and should return to a "safe" spot.

@winder... I have read that documentation, but in practice issuing G43.1 doesn't seem to affect anything. After every probe, I still have to manually issue a G92 command, or my work "zero" doesn't get set correctly.

vMeph, 2017-04-24:
will be better if we set the height were we want to retract Z, will be a more complete action....

cjbaar, 2017-04-25:
OK, well let me retract my earlier statement. I have been using the probe more, and now it does seem to be resting the z-axis properly. Not sure why it didn't seem to do it before, but the current method is working for me. Adding a back-off would still be a nice touch. :)

cjbaar, 2017-04-25:
Nope. Spoke too soon again.  :)  I restarted UGS, and now the probe is not setting Z height correctly again.

cjbaar, 2017-04-25:
OK... a bit more rambling. :)
It appears that the first use of the probe after restarting UGS or re-connecting the serial port does not reset the z-axis to where it needs to be. I am still issuing the "G92 Zn" command. However, after I have done this once, subsequent uses of the probe *do* properly reset the z-axis with only the use of G43.1. But it does appear, to me anyway, that G92 does play an important role.

winder, 2017-04-25:
@cjbaar have you tried using g43.1 instead of g92? The documentation I've read always suggests that g43.1 is the preferred method.

Can you identify anything else about your workflow that could cause this?
Are you probing right after restarting grbl?
After homing / jogging to a probe location?
What version of grbl are you using?

Here is some additional documentation https://www.shapeoko.com/wiki/index.php/Touch_Plate

cjbaar, 2017-04-25:
Well, maybe I am not using the zero function correctly. However, I find that my jobs do not end up in the correct place unless the tip of the bit, on the surface, is showing Z 0.00 before any run. So I am re-running the probe every time I change the tool, or every time UGS forces me to restart and I have to redo the whole homing sequence. My grbl firmware is 0.9j

As I have discovered (most recent post), using the probe sequence with the programmed tool height command (43.1) after changing tools does appear to work correctly. However, for me, the first time it is run after any connect or restart, it has no reference point and therefore needs the G92 to be zeroed correctly.

kulpio, 2019-04-15:
Hi everyone,
First off all, thank you @winder for your awesome sender, awesome job.
I don't know if I missed something but I am running into issues setting new tool for jobs. 
What I do is export from Fusion operations by operations for every tool changes, but since each of my tools have different length, it is hard to position perfectly to all the same Z height.
I do not have a z-probe, but I could buy one.
My questions are:
Can I z-probe for each tool? Does setting the tool length in Fusion changes anything? (should I set them to the same height for all so I have a constant z-zero?
Any input is appreciated.

PS: I use a Shapeoko XL with upgraded z-axis from CNC4Newbie with a Makita spindle (I might upgrade to a water cooled 2kw spindle).

Best,

d.

jahnj0584, 2019-04-15:
The probes are on eBay for about 6$.

You just reset G54 for each tool off the WCS chosen in your Setup and hit
cycle start. Easy.

carneeki, 2019-04-15:
A touch plate might be what you're looking for?

Probes are good for X and Y axes, but for Z, if your tools don't have a
preset length or stop (it sounds like yours do not), a probe probably won't
help you for Z. (assuming a mill/router, for lathes, a QCTP is pretty much
the only way to go!)

A touch plate / block uses the probe pin on the controller, and the cutter
itself as a path to ground. The plate is placed atop the work, and when the
(stationary) cutter touches the plate, the probe pin is grounded through
the cutter, and the controller stops motion, from here, you just input the
thickness of the plate as your Z offset, move the plate away, and you're
done. No more tool changes, and no need to fiddle with measuring the tool
length - it works for all tools in the cutter as long as the cutter is
grounded (the machine itself is generally grounded).

Commercial units can be purchased, or something as simple as a piece of
steel (or even PCB) with a wire attached to it depending on your needs...
If you DIY one, ensure that the bottom of the plate is shielded from the
machine / work, or else the probe pin will be grounded through the work
rather than the cutter. Commercial units are generally two pieces of steel
sandwiched around some plastic and a sprung button to a known (and slightly
adjustable) height. But a PCB is easily mic'ed and used as Z offset too,
and it is also cheap to prototype a proof of concept. The internet is full
of plans and builds, all variations on the same theme. As with your limit
switches and input buttons, ensure you filter the inputs.

Good luck!

kulpio, 2019-04-15:
Thank you everyone. I purchased a cheap Z touch probe, I think it will do plenty for what I need. I might DIY one for xyz, but not now .)
Thanks!

kulpio, 2019-04-18:
Just got my cheap $6 probe and it works like a charm with the probe module. Thank you everyone for the help. Thanks Winder for the making a top sender.