winder/Universal-G-Code-Sender #589: GC: initialisation line

71tr6, 2017-02-26:
Is there a way to edit this line?
Most of my work is inches so I'd like to change the g21 to a g20.

winder, 2017-02-27:
I'm not sure what you're asking, shouldn't your CAM software should be able to generate code in inch/mm? Or are you asking about the units in the UGS machine status area?

71tr6, 2017-02-27:
In the console, after a homing operation, you get an initialization line much like CAM software generates at the beginning of a file.
Specifically: [GC: G0 G54 G17 G21 G90 G94 M5 M9 T0 F0.0 S0]
I'd like to be able to change the G21 to G20.  Or, even better, have it look at the jogging window and use the same units that I'm working in.

vMeph, 2017-02-27:
You set that on grbl configs $$

71tr6, 2017-02-27:
vMeph, if you're referring to the $13 value, I do have it set to 1. 
The $13 value is where I would LIKE the G20/G21 value to come from.
Is this a bug or a wish list item?

vMeph, 2017-02-27:
You have to set $13=0 for mm

winder, 2017-02-27:
`$13` is how GRBL reports units. I'm not sure how the initial units gets set - it may be hard coded to start in mm.

You can put a `G20` command in a startup line: https://github.com/gnea/grbl/blob/f51268e8556354a4851a8bdac2b6be7b8d7d5a03/doc/markdown/interface.md#startup-line-execution

closed 2017-02-27 by winder

winder, 2017-02-27:
Here's a better link about the startup commands:
https://github.com/gnea/grbl/blob/e455764079ccfa9ae277a8fa0078b0dbb36f3a51/doc/markdown/commands.md#n---view-startup-blocks