winder/Universal-G-Code-Sender #590: Option to Turn Tool Off When Paused

rmisko11, 2017-02-28:
Maybe this option exists buy I can't find it so I would to suggest that there be an option to stop the tool, e.g. M5 when pausing. Then reset the tool state upon resume.  I am using a laser engraver and often I pause while laser is on which typically burns a hole in the target as it does not automatically turn off.

winder, 2017-02-28:
What version of GRBL are you using? v1.1 should do this for you automatically.

rmisko11, 2017-02-28:
I tried with the nightly builds of both classic gui and platform and it didn't work for me in either case.

rmisko11, 2017-02-28:
Sorry, I don't read well and just noticed you are talking about GRBL.  The laser engraver has version 0.9 installed.

winder, 2017-02-28:
@rmisko11 unfortunately there isn't a very good way for UGS to implement this functionality. But GRBL v1.1 has a lot of features (including turning off the laser during a feed hold), so it may be worth an upgrade. Here is some documentation: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

closed 2017-02-28 by winder

rmisko11, 2017-02-28:
Will,

Thank you for your prompt responses.  GRBL Version 1.1. looks pretty nice.
I will check if they engraver manufacturer has a new release but I am
guessing not.  I think it would be difficult for me to customize GRBL
source and compile for an arduino.  It is quite easy to build a simple
sender that would pause and resume the way I want which I would probably
have to do in any event.  At this time the only reason I want to do that is
for long prints because the engraver manufacturer suggests that the laser
lasts longer if it is turned off for half hour every hour to 'cool down'.
Seems an excessively long cool down but generally it sounds good to pause
intermittently for the laser to cool.

Even so, I will likely be using your UGS Platform.  Only started using it
today and it seems really good.

rmisko11, 2017-02-28:
Thanks for the suggestion.  I found the source for GRBL1.1f ran the
makefile then installed and it seems to work!  Took about 2 mins!  I had no
idea it was that easy.

I will have to fine tune the settings but otherwise it seems quite
compatible with your UGS Platform.

Thanks again.

rmisko11, 2017-02-28:
> @rmisko11 <https://github.com/rmisko11> great! Thanks for the update. UGS
> Platform introduced a plugin system, it should be fairly simple to make a
> plugin to pause a job after 30 minutes of continuous use if you're so
> inclined. This document is a bit outdated, but the code is still accurate:
> http://winder.github.io/ugs_website/dev/plugin/

Thanks, I will have a look at that.  I find that the 'Pause'  in GRBL v1.1f
 does not always turn off the laser and I do have $32=1.  So if I can
create a plugin to pause then I should also be able to have it send either
M5 or M3 as required.

winder, 2017-02-28:
@rmisko11 The problem with issuing a `M5` or `M3` is that is that a GRBL pause (a.k.a. feed hold) usually stops the machine half way through a command, and when holding it doesn't accept additional commands.

In UGS Platform under `Window > Overrides` to toggle the spindle in "real time" which you can try using.

I'm surprised that grbl doesn't disable the laser. Take a close look at that laser-mode documentation, though, Chamnit goes into really nice detail about how things should behave - and create a bug if you find a discrepancy between what you read and the behavior you see. This is a new feature after all so there are probably a couple edge cases waiting to be found.