winder/Universal-G-Code-Sender #594: Jog Controls Locked out

aforww, 2017-03-03:
If I send a job and I cancel it, or it completes, all jog functionality is locked out until I close and reopen UGS-P. Utilizing GRBL 1.1f as well.

winder, 2017-03-04:
@aforww In UGS-P there is an "About ugsplatform" page in the menu, could you copy your system information from that page and put it here?

I'm not able to reproduce this problem. To double check I just updated to the latest GRBL 1.1f, started a send and then canceled it.

aforww, 2017-03-04:
@winder After you said you couldn't duplicate the issue I deleted UGS-P and downloaded it again. Wouldn't you know it, No more issues. The version I had was only a few days old so guess it was something that was worked out. Thanks.
ow it, No more issue. The version I had was only a few days old so guess it was something that was worked out. Thanks.

closed 2017-03-04 by aforww