winder/Universal-G-Code-Sender #595: Problem activate macro with gamepad

ermendaa, 2017-03-03:
The problem or bug I have is the following; **When I activate the macro with the gamepad I only execute the first order of the macro, the rest ignores it**. A greeting and thanks for your work.

winder, 2017-03-04:
@ermendaa What is your macro?

ermendaa, 2017-03-04:
Is this: $H; G10 P0 L20 X0 Y0 Z0; G30.1 X-175 Y-255 Z-2 F100; G0 Z-2;G10 P0 L20 X0 Y0 Z0

winder, 2017-03-08:
I think this is probably a side effect of how `$H` works. If you type these commands in manually do things get ignored while the machine is homing?

rjclocks, 2017-03-13:
I am also having macro issues - only the first command in a macro chain is being processed.
Running nightly UGS platform from 11th March on a Mac.
Macro is only able to be executed from the menu, it won't respond from the window buttons, but still only executes the first command in the string.
Macro: G91; G38.2 Z-10 F100

rjclocks, 2017-03-13:
Update - My macro issues have been resolved by updating my OS to the latest version. Not sure why, but all are behaving as expected.

winder, 2017-03-30:
@ermendaa I found the problem, thanks for the bug report.

commit 4f4e3230 (2017-03-30): Fix macro processing in Platform menu. Closes #595
closed 2017-03-30 by winder