winder/Universal-G-Code-Sender #596: G91 Not Working in 2017-02-19 UGS Platform

rmisko11, 2017-03-05:
I can't get relative coordinates (G90) to work;

I have attached the screen capture which should show the movement for a rectangle that ends up at (0,0).  There should be no negative X or Y coordinates.

[image]

The code shown works as I expect when I send it directly to Laser Engraver.

jahnj0584, 2017-03-05:
It says in the code that it switches to G91 then back to G90 after its done

rmisko11, 2017-03-05:
That's true but the code in between should moving the laser according to relative coordinates which it is not

winder, 2017-03-08:
@rmisko11 Was `G91` in that command remover filter we found the other day? In the visualizer the lines are blocked by the X/Y axis lines, but otherwise looks good to me. It makes a 100mm angle:
[image]

rmisko11, 2017-03-08:
Sorry I forgot about this issue.  Yes, it was in the filter and since
removing it G91 works perfectly.

closed 2017-03-08 by winder