winder/Universal-G-Code-Sender #597: Can not connect to cnc card "Grbl 0.9j ['$' for help] <Idle, WPos> gives an error.

Seyko5, 2017-03-06:
Can not connect to cnc card "Grbl 0.9j ['$' for help]
<Idle, WPos> gives an error.

I want a solution

Seyko5, 2017-03-06:
[image]

rmisko11, 2017-03-06:
It shows that you ARE connected.

Seyko5, 2017-03-06:
The stepper motors are not moving. Look at the picture

Android için Outlook<https://aka.ms/ghei36> uygulamasını edinin

Seyko5, 2017-03-06:
The stepper motors are not moving. Look at the picture

winder, 2017-03-06:
@Seyko5 Are you resetting your GRBL board in your screenshot? You shouldn't see `Grbl 0.9j ['$' for help]` over and over again. Also the `<Idle, WPos` showing up is very unusual, it means the `>` is not being received from GRBL. What do you see if you select `Show verbose output`?

Are you using a GRBL clone? There are some known issues with CH340G based boards: https://github.com/grbl/grbl/wiki/Known-Bugs#usb-to-serial-transmission-errors

Also, thank you for the Turkish translation. I'll update UGS soon to include it in the nightly build.

closed 2017-03-12 by winder