winder/Universal-G-Code-Sender #598: UGS Platform Ignoring Feed Rate

rmisko11, 2017-03-06:
UGS Platform (Build 19 Feb 17) is ignoring the speed command, say, 'F100'.  As can be seen in attached image the source shows 'F5000' at line 28 but it can be seen in the Console that it was not sent to the Laser Engraver.  It should have been right after the third line in the Console.

I have used software reset, program restart, and even $X to no avail.

Note: The program works when sent using my UGS Sender.  

[image]

jahnj0584, 2017-03-06:
Is your sent feed rate greater then the Max speed in your $$ settings??

rmisko11, 2017-03-06:
No, they are set to 5000.

winder, 2017-03-06:
Very strange, running it on the current nightly build I see the F command go through:

```
>>> G21
>>> G90
>>> G10P0L20X0Y0Z0
>>> M4
>>> S12
>>> F5000
>>> G0Y0
>>> G0X0
>>> G0X10
ok
ok
ok
ok
ok
ok
ok
ok

**** Finished sending file in 00:00:00 ****

ok
```

Is there a chance you were editing the file after loading it? UGS doesn't automatically reload the file if you make changes.

winder, 2017-03-06:
@rmisko11 Could you take a screenshot of the UGS `Controller Options` preferences?

rmisko11, 2017-03-06:
I reload the file after each edit.  The same file sent not using the
Platform works fine and I am sure F5000 used to work on this platform on my
computer.

rmisko11, 2017-03-06:
This is what I get when tried then same snippet:
**** Connected to /dev/ttyUSB0 @ 115200 baud ****

Grbl 1.1f ['$' for help]
>>> $$
>>> $G
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 10    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 1    (Laser-mode enable, boolean)
$100 = 80.000    (X-axis travel resolution, step/mm)
$101 = 80.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 5000.000    (X-axis maximum rate, mm/min)
$111 = 5000.000    (Y-axis maximum rate, mm/min)
$112 = 50.000    (Z-axis maximum rate, mm/min)
$120 = 500.000    (X-axis acceleration, mm/sec^2)
$121 = 500.000    (Y-axis acceleration, mm/sec^2)
$122 = 500.000    (Z-axis acceleration, mm/sec^2)
$130 = 125.000    (X-axis maximum travel, millimeters)
$131 = 190.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
Skipping command #2
>>> G90
>>> G10P0L20X0Y0Z0
>>> M4
>>> S12
Skipping command #7
Skipping command #8
>>> G0X0
>>> G0X10
Skipping command #11
ok
ok
ok
ok
ok
ok

Are the options at the top what you wanted?

winder, 2017-03-07:
@rmisko11 I was looking for the page at the bottom here: http://winder.github.io/ugs_website/guide/common/#error-handling

rmisko11, 2017-03-07:
> handling

Ahhh, thank you!  That fixed it.  I saw that there were a bunch of commands
in "Regular Expression Pattern Remover" that I then deleted.

UGS Platform is now working as I expect.

I am happy to have the ticket closed.

Thanks,
  Roger

closed 2017-03-07 by winder