winder/Universal-G-Code-Sender #599: UGS no serial ports found

isitcrunchy, 2017-03-06:
I also have the gcode-sender Chrome app working on my macbook and it finds my serial port just fine:

[image]

My macbook knows the serial port is there

[image]

But UGS can't find it

[image]

I have the Eleks mill from china
It has the chinese auridino that needed the patch from
/dev/tty.wchusbserial1420

I have tried turning everything on and off multiple ways

Any help would be appreciated

winder, 2017-03-06:
You can paste that into the port text box to connect. There's a bug in the
serial library I'm using that causes this.

isitcrunchy, 2017-03-06:
I must have tried pasting that in a dozen times. But now, this time, with your help it worked.

Thanks for waving whatever magic wand you have.

closed 2017-03-07 by winder

vleroylr, 2018-02-27:
Hi, I'm having the same problem, what was it that you pasted and solved the problem? I have windows 10

luiscabus, 2018-03-19:
I am having the same problem. Only pasting /dev/tty.wchusbserial1420 makes it work

stevecooley, 2018-03-21:
I am also having this problem with a woodpecker 3.0.1 board, and GRBL 0.9