winder/Universal-G-Code-Sender #600: Can't read/send file on Platform on a Mac

justflie, 2017-03-11:
I've tried this on two different macs thus far with the same result. I was first attempting to send a file via platform on an old Windows laptop. It simply wasn't up to the task for a larger file (pegging the processor, eventually leading to issues in the cut). I switched over to a Mac to try. 

When I load the file, nothing shows up in visualizer. When I attempt to send the file, it gives me a null error as if there's nothing to send. The same file works fine in UGS 1.0.9 on the same Mac and loaded fine on the windows machine. Running the latest 1.8.x java and the latest nightly build of UGS platform. I've tried several different files and I can't get any to load/send from platform on two different Macs. Am I doing something stupid? 

I've never put an issue in GitHub before so please tell me what kind of debugging information you need and I will provide it. Sorry for the bother.

winder, 2017-03-11:
@justflie could you share the file you're using?

In the menu under "About ugsplatform", could you copy and paste the version information at the bottom?

For example mine says this:
```
Product Version: ugs-platform 201510222201
Java: 1.8.0_60; Java HotSpot(TM) 64-Bit Server VM 25.60-b23
Runtime: Java(TM) SE Runtime Environment 1.8.0_60-b27
System: Mac OS X version 10.11.6 running on x86_64; UTF-8; en_US (ugsplatform)
User directory: /opt/Universal-G-Code-Sender/ugs-platform/application/target/userdir
Cache directory: /opt/Universal-G-Code-Sender/ugs-platform/application/target/userdir/var/cache
```

justflie, 2017-03-11:
Sure thing. The first file I tried was the larger finish one. Then I tried the simple rough one and got the same (no) result. Thanks, Winder.
[Archive.zip](https://github.com/winder/Universal-G-Code-Sender/files/836205/Archive.zip)

Product Version: NetBeans Platform 8.1 (Build 201510222201)
Java: 1.8.0_121; Java HotSpot(TM) 64-Bit Server VM 25.121-b13
Runtime: Java(TM) SE Runtime Environment 1.8.0_121-b13
System: Mac OS X version 10.12.3 running on x86_64; UTF-8; en_US (serverlaunch)
User directory: /Users/George/Library/Application Support/ugsplatform/dev
Cache directory: /Users/George/Library/Application Support/ugsplatform/dev/var/cache

winder, 2017-03-12:
So good news is that your file loads up fine on my mac test machine. I updated java to v121 and still no problems.

You mentioned running the nightly build, is the visualizer the only part not working for you right now?

When you see the problem, is there a red error icon in the bottom status area?

justflie, 2017-03-13:
Good news! 

When I attempt to load the file in platform, it appears to load (as in, the green start key lights up), but nothing appears in visualizer. When I attempt to send the file, I receive the following error: Error while starting file stream: null (screenshot attached)
[image]
. The same error happens when I attempt to run any other file in Platform on this or my other Mac.

I've run the same file successfully (with visualizer working as expected) in the stable 1.0.9 release and the 2.0 nightly build. I'm just having an issue with Platform. Is it possible there is some configuration step with Platform I've missed? My understanding was i just need to double click the file in bin, terminal fires up, and we're off and running. 

PS: I know the rest of my settings are fubard. I was just sending the file to get the error screenshot so i didn't bother moving to the work zero, etc.

rjclocks, 2017-03-13:
I am seeing the same problem with the Platform 2.0 nightly (11th Mar) on Mac
```
Product Version: NetBeans Platform 8.1 (Build 201510222201)
Java: 1.8.0_121; Java HotSpot(TM) 64-Bit Server VM 25.121-b13
Runtime: Java(TM) SE Runtime Environment 1.8.0_121-b13
System: Mac OS X version 10.9.5 running on x86_64; UTF-8; en_US (serverlaunch)
User directory: /Users/R/Library/Application Support/ugsplatform/dev
Cache directory: /Users/R/Library/Application Support/ugsplatform/dev/var/cache
```
I have tried different gcode files and am seeing the same issue. The terminal window is showing that the file is being processed (bounds and line count), but something is not happening after that.
```
Object bounds: X (-55.5, 55.5)
               Y (-66.0, 66.0)
               Z (-18.0, 15.0)
Center = (0.0, 0.0, -1.5)
Num Line Segments :2538
```

rjclocks, 2017-03-13:
Just looked for the red error icon in the status bar and found the following:
```
java.lang.NoSuchMethodError: javax.vecmath.Point3d.setX(D)V
	at com.willwinder.ugs.nbm.visualizer.renderables.GcodeModel.generateObject(GcodeModel.java:255)
	at com.willwinder.ugs.nbm.visualizer.renderables.GcodeModel.setFile(GcodeModel.java:124)
	at com.willwinder.ugs.nbm.visualizer.renderables.GcodeModel.setGcodeFile(GcodeModel.java:103)
	at com.willwinder.ugs.nbm.visualizer.RendererInputHandler.setGcodeFile(RendererInputHandler.java:98)
	at com.willwinder.ugs.nbm.visualizer.RendererInputHandler.UGSEvent(RendererInputHandler.java:117)
	at com.willwinder.universalgcodesender.model.GUIBackend.sendControlStateEvent(GUIBackend.java:905)
	at com.willwinder.universalgcodesender.model.GUIBackend.setGcodeFile(GUIBackend.java:465)
	at com.willwinder.universalgcodesender.uielements.actions.OpenGcodeFileAction.openGcodeFile(OpenGcodeFileAction.java:49)
	at com.willwinder.universalgcodesender.uielements.actions.OpenGcodeFileAction.openGcodeFileDialog(OpenGcodeFileAction.java:65)
	at com.willwinder.universalgcodesender.uielements.actions.OpenGcodeFileAction.actionPerformed(OpenGcodeFileAction.java:44)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
[catch] at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

winder, 2017-03-13:
@rjclocks Thanks, that was what I needed.

@justflie @rjclocks I updated the nightly build, could you give it a try and see if the issue is resolved?

rjclocks, 2017-03-13:
Update - My file sending issues have been resolved by updating my OS to the latest version. Not sure why, but all is now behaving as expected. 
I will grab the new nightly build. Any tips on what to backup so as not to have to reconfigure everything?

winder, 2017-03-13:
@rjclocks These directories you shared earlier have all the platform metadata:
```
User directory: /Users/R/Library/Application Support/ugsplatform/dev
Cache directory: /Users/R/Library/Application Support/ugsplatform/dev/var/cache
```

The problem you posted earlier, `java.lang.NoSuchMethodError: javax.vecmath.Point3d.setX(D)`, is most likely due to a different version of that library being on your machine. So it's feasible that an OS update could fix it. The nightly build should also fix the problem even if the wrong library is loaded.

justflie, 2017-03-14:
Good here, Winder. That did the trick. Thanks.

closed 2017-03-14 by winder