winder/Universal-G-Code-Sender #601: Platform alarm reset? How to do it.

aforww, 2017-03-16:
When I am using soft limits and the machine activates the alarm, how do I reset it? I tried $x with no success. The jog control buttons, machine control buttons, and manually entered code didn't work unless I completely close UGS and open it again .

labohemia, 2017-03-29:
you Need to write $X the X is big 
if you write $x it just saves your G-Code Setting

cjbaar, 2017-04-25:
I second this one. When soft limits are hit, my "Common Actions" panel grays out and never returns. Issuing a manual "$X" does nothing. I have to quit and restart UGS.

vMeph, 2017-04-25:
when grbl issues a alarm you have always to reset arduino

cjbaar, 2017-04-25:
I though the arduino only needed to be reset on hard limits, not soft ones. But, in either case, I shouldn't have to quit restart UGS. This isn't an issue with the arduino not responding, it's one of UGS not allowing access to buttons.

cjbaar, 2017-05-16:
I'd like to bump this up again. Would be nice to recover from errors/alarms without having to restart UGS

allcustomz, 2017-06-27:
I got same issue

aforww, 2017-07-08:
I was really hoping to see this in 2.0 lol. But in all reality, it's a minor annoyance compared to all the Amazing work he has put into it lately. Fingers crossed for the future!

winder, 2017-07-09:
Duplicate of #563

closed 2017-07-09 by winder