winder/Universal-G-Code-Sender #602: Schedule a file task

jmesar, 2017-03-18:
I need that my CNC execute a gcode file at certain time in the day hh:mm:ss is that possible?

My need is simple, the task is so long , Aspide say 12h, that i have to split it in several pieces, and they have to be done "Stand Alone" while i am at the office.  several small  pieces time my neighbors do not complain by the noise.

jahnj0584, 2017-03-18:
Set up a webcam stream and remote desktop. This isn't the safest thing to
do. Are you sure your speeds are set properly for it to take that long

<https://github.com/winder/Universal-G-Code-Sender/issues/602>, or mute the
thread

winder, 2017-03-31:
Sorry but I'm not planning to add any features like this, I don't like the idea of encouraging people to run their machines unattended.

closed 2017-03-31 by winder