winder/Universal-G-Code-Sender #603: An error was detected while sending (only on nightly builds)

FishyD, 2017-03-19:
Stable classic UGS (1.09) works great but running the same G-Code with both classic and platform  nightlies are giving me the same errors. When I load any G-Code file I get "an error was detected while sending..." If I try to run the Triquentra-CNC probe file I get a G20 error. If I run a V-Carve file w/ the X-Carve PP I get a M6 error. I know GRBL has issues with the M6 command so I could just delete that or program UGS to ignore it but why does 1.09 not give me the same error while running the identical G-Code? 1.09 runs both of these G-Codes flawlessly. See below for 2 screen shots. 

Is there some additional setting I'm missing in the nightlies that are enabled by default in the stable versions? 

The Triquentra CNC code giving me issues is: 
G20
G92 X0
G92 Y0
G92 Z0
G38.2 X-0.5F2
G92 X2.0375
G91 G0 X0.5
G91 G0 Y1
G91 G0 X-1
G38.2 Y-1 F2
G92 Y2.0925
G91 G0 Y0.125
G91 G0 Z1.019
G91 G0 Y-0.625
G38.2 Z-1 F3
G91 G0 Z0.05
G38.2 Z-0.2 F1
G92 Z0.8210
G91 G0 Z0.125
G91 G0 X1
G91 G0 Y1
M02

[image]
[image]

winder, 2017-03-20:
@FishyD This is a new feature in the nightly build based on the recommended way to interact with GRBL. In UGS 1.0.9 those commands still failed, but you aren't notified of the failure.

The first failure is because there are some non-ascii characters in the file, if you remake the file it should work fine. In this case the file is setting the units, so it's really important that the user is notified that there was a problem!

The second failure is because GRBL doesn't support `M6`, In this case choosing `Yes` to skip the unsupported command is appropriate.

Let me know if you have any other questions.

closed 2017-03-20 by winder

FishyD, 2017-04-01:
Thanks! Regarding the Triquentra probe files; for some reason the .txt file was saving as a UTF-8 txt file. I opened it, resaved as an ANSI txt file and it worked great. I'll have to talk to Charlie on getting his Tool Box to save the txt output files as ANSI by default. On the other error I removed the "T1M6" from my post processor and no errors. Thanks for the help, I really appreciate it.

carneeki, 2018-04-19:
> @gwag18 <https://github.com/gwag18> please create a new issue and share
> your gcode file so that we can take a look.

At first glance, I'm not sure this is an error with UGS, but maybe the post
processor in Inkscape.

'OS30' (or '0S30') are not valid G-codes for GRBL, before posting the
g-code file, can you check around line 16 or 20ish and confirm that the
invalid codes aren't generated by Inkscape?

gwag18, 2018-04-21:
Thank you for answer. You should not re-create files created in inkscape. I still get the same error message. Send the corresponding g-code below.

(Scribbled version of C:\Users\bsgwa\AppData\Local\Temp\ink_ext_XXXXXX.svgTE4HHZ @ 3500.00)
( unicorn.py --tab="plotter_setup" --pen-up-angle=50 --pen-down-angle=30 --start-delay=150 --stop-delay=150 --xy-feedrate=3500 --z-feedrate=150 --z-height=0 --finished-height=0 --register-pen=true --x-home=0 --y-home=0 --num-copies=1 --continuous=false --pause-on-layer-change=false C:\Users\bsgwa\AppData\Local\Temp\ink_ext_XXXXXX.svgTE4HHZ )
G21 (metric ftw)
G90 (absolute mode)
G92 X0.00 Y0.00 Z0.00 (you are here)

M300 S30 (pen down)
G4 P150 (wait 150ms)
M300 S50 (pen up)
G4 P150 (wait 150ms)
M18 (disengage drives)
M01 (Was registration test successful?)
M17 (engage drives if YES, and continue)

(Polyline consisting of 1 segments.)
G1 X1.67 Y2.09 F3500.00
M300 S30.00 (pen down)
G4 P150 (wait 150ms)
G1 X1.21 Y2.27 F3500.00
G1 X1.12 Y2.72 F3500.00
G1 X1.58 Y3.61 F3500.00
G1 X2.51 Y3.92 F3500.00
G1 X3.42 Y3.69 F3500.00
G1 X4.20 Y3.08 F3500.00
G1 X4.71 Y2.26 F3500.00
G1 X4.88 Y0.72 F3500.00
G1 X4.35 Y-0.68 F3500.00
G1 X3.30 Y-1.79 F3500.00
G1 X1.92 Y-2.47 F3500.00
G1 X-0.22 Y-2.53 F3500.00
G1 X-2.11 Y-1.70 F3500.00
G1 X-3.58 Y-0.20 F3500.00
G1 X-4.42 Y1.76 F3500.00
G1 X-4.55 Y3.15 F3500.00
G1 X-4.38 Y4.49 F3500.00
G1 X-3.25 Y6.88 F3500.00
G1 X-1.28 Y8.70 F3500.00
G1 X1.25 Y9.70 F3500.00
G1 X2.95 Y9.82 F3500.00
G1 X4.57 Y9.57 F3500.00
G1 X6.09 Y9.00 F3500.00
G1 X7.47 Y8.13 F3500.00
G1 X8.66 Y7.03 F3500.00
G1 X9.64 Y5.71 F3500.00
G1 X10.36 Y4.22 F3500.00
G1 X10.79 Y2.59 F3500.00
G1 X10.89 Y0.59 F3500.00
G1 X10.57 Y-1.32 F3500.00
G1 X9.87 Y-3.11 F3500.00
G1 X8.83 Y-4.72 F3500.00
G1 X7.51 Y-6.11 F3500.00
G1 X5.94 Y-7.24 F3500.00
G1 X4.17 Y-8.07 F3500.00
G1 X2.25 Y-8.55 F3500.00
G1 X-0.06 Y-8.63 F3500.00
G1 X-2.26 Y-8.24 F3500.00
G1 X-4.31 Y-7.41 F3500.00
G1 X-6.15 Y-6.20 F3500.00
G1 X-7.75 Y-4.66 F3500.00
G1 X-9.03 Y-2.84 F3500.00
G1 X-9.97 Y-0.80 F3500.00
G1 X-10.50 Y1.43 F3500.00
G1 X-10.57 Y4.04 F3500.00
G1 X-10.09 Y6.53 F3500.00
G1 X-9.14 Y8.85 F3500.00
G1 X-7.75 Y10.93 F3500.00
G1 X-6.00 Y12.72 F3500.00
G1 X-3.93 Y14.16 F3500.00
G1 X-1.61 Y15.20 F3500.00
G1 X0.92 Y15.78 F3500.00
G1 X3.83 Y15.83 F3500.00
G1 X6.61 Y15.28 F3500.00
G1 X9.19 Y14.20 F3500.00
G1 X11.51 Y12.64 F3500.00
G1 X13.50 Y10.67 F3500.00
G1 X15.10 Y8.36 F3500.00
G1 X16.24 Y5.75 F3500.00
G1 X16.87 Y2.93 F3500.00
G1 X16.91 Y-0.29 F3500.00
G1 X16.29 Y-3.36 F3500.00
G1 X15.07 Y-6.21 F3500.00
G1 X13.34 Y-8.76 F3500.00
G1 X11.16 Y-10.95 F3500.00
G1 X8.59 Y-12.70 F3500.00
G1 X5.71 Y-13.96 F3500.00
G1 X2.58 Y-14.63 F3500.00
G1 X-0.94 Y-14.65 F3500.00
G1 X-4.30 Y-13.96 F3500.00
G1 X-7.41 Y-12.62 F3500.00
G1 X-10.20 Y-10.71 F3500.00
G1 X-12.59 Y-8.31 F3500.00
G1 X-14.50 Y-5.49 F3500.00
G1 X-15.85 Y-2.33 F3500.00
G1 X-16.58 Y1.10 F3500.00
M300 S50.00 (pen up)
G4 P150 (wait 150ms)

(end of print job)
M300 S50.00 (pen up)
G4 P150 (wait 150ms)
M300 S255 (turn off servo)
G1 X0 Y0 F3500.00
G1 Z0.00 F150.00 (go up to finished level)
G1 X0.00 Y0.00 F3500.00 (go home)
M18 (drives off)

carneeki, 2018-04-22:
I don't think this is an error with UGS.

What's going on:
M300 is not a code supported by GRBL (the controller), so GRBL returns the
error. UGS simply acts as an interface, it passes the illegal code on to
GRBL, and passes the error back from GRBL to the you the user.

How to fix it:
1) This isn't really the right place for detailed advice on Inkscape. Best
to try an Inkscape+GRBL forum / message board.

2) That said, what you're wanting to do is get Inkscape to produce G-code
that is compatible with GRBL. Checkout this video:

https://www.youtube.com/watch?v=rLBU9XjDB9U around 5m20 here we can see the
programmer configures Inkscape to use "M3" for up (I speak 'just enough
Hebrew' to recognise the numbers) and "M5" for down, and then uses notepad
to search for M3 and replace it with "Z-1" and M5 with "Z1". He could
probably just use "G1 Z1" and "G1 Z-1" in the window settings and avoid the
find+replace.

In your case, it looks like you send M300 S30 for down and M300 S50 for up.
I would consider changing this to G1 Z0 and G1 Z10 (changing Z10 and Z0 for
numbers that are suitable for your machine). G1 is an instruction to tell
the machine to move in a line. Z means the Z axis (up+down), and 10 is the
co-ordinate or distance to move to.

If it's the first time running GRBL, I'd pull the tool or pen out of the
spindle incase it crashes into the work. Be ready to emergency stop the
machine when it starts.

Good luck!

gwag18, 2018-04-22:
Thank you for your mail. The code was not really gcode. I have not been able to insert MI GRBL in the Extensions of inkscape. Now gcode is created and sent.
Let me ask you one more question. Where should I connect the servo motor jumper cable on the cnc shield? I would like to receive your help.

Windows 10용 메일에서 보냄

보낸 사람: Adam
보낸 날짜: 2018년 4월 22일 일요일 오후 12:35
받는 사람: winder/Universal-G-Code-Sender
참조: byung se gwag; Mention
제목: Re: [winder/Universal-G-Code-Sender] An error was detected whilesending (only on nightly builds) (#603)

I don't think this is an error with UGS.

What's going on:
M300 is not a code supported by GRBL (the controller), so GRBL returns the
error. UGS simply acts as an interface, it passes the illegal code on to
GRBL, and passes the error back from GRBL to the you the user.

How to fix it:
1) This isn't really the right place for detailed advice on Inkscape. Best
to try an Inkscape+GRBL forum / message board.

2) That said, what you're wanting to do is get Inkscape to produce G-code
that is compatible with GRBL. Checkout this video:

https://www.youtube.com/watch?v=rLBU9XjDB9U around 5m20 here we can see the
programmer configures Inkscape to use "M3" for up (I speak 'just enough
Hebrew' to recognise the numbers) and "M5" for down, and then uses notepad
to search for M3 and replace it with "Z-1" and M5 with "Z1". He could
probably just use "G1 Z1" and "G1 Z-1" in the window settings and avoid the
find+replace.

In your case, it looks like you send M300 S30 for down and M300 S50 for up.
I would consider changing this to G1 Z0 and G1 Z10 (changing Z10 and Z0 for
numbers that are suitable for your machine). G1 is an instruction to tell
the machine to move in a line. Z means the Z axis (up+down), and 10 is the
co-ordinate or distance to move to.

If it's the first time running GRBL, I'd pull the tool or pen out of the
spindle incase it crashes into the work. Be ready to emergency stop the
machine when it starts.

Good luck!

—

---
AVG에서 이 이메일에 대한 바이러스 검사를 완료하였습니다.
http://www.avg.com

carneeki, 2018-04-23:
Best to read the manual for your shield, this is because there are a lot of
different CNC shields, and their connection types differ from shield to
shield.

Usually they are a larger plastic connector with 4 or more wires for each
motor and will typically be labelled for each axis on the machine (eg XYZ,
and sometimes XXYZ or XYYZ [where one axis is cloned on bigger machines],
or XYZA for machines that have a rotary table [standard GRBL does not have
this]).

Good luck!

winder, 2018-04-23:
Please create a new issue if there are further questions.