winder/Universal-G-Code-Sender #604: UGS platform G54 problems?

vMeph, 2017-03-19:
the problem is, lets say i do this
i do a homing $H lets say will stay at X-5 Y-5  Z-5 due to pull off distance
i send machine to a position lets say X-100 Y-100
i hit G54(G10 P1 L20 X0 Y0 Z0)
when i do this the machine coordinates becomes positive X100 Y100 and work coordinates X0 Y0
so when i run a job the machine will add to the machine positive positions other then discount from machine coords

it seems that there is a bug on software cause when you send a G54 the machine coordinates will invert to positive coords
any ideas?

jahnj0584, 2017-03-19:
You're home position should still end as 0,0,0 and moving right and away
should be positive values

vMeph, 2017-03-19:
dont get it if machine is 0,0,0 and i move negative to -100,-100,0  and hit g54 and machine becomes positive 100,100,0 and work position 0,0,0 if i send home again and machine become 200,200??

jahnj0584, 2017-03-19:
sounds like you have absolute and relative coordinate systems mixed up.

vMeph, 2017-03-19:
i have reflash arduino still same problem could you do a test dont know what UGS platform but i have the 727

vMeph, 2017-03-19:
if you are at machine coordinates-50,-50,0 when you hit G54 it becomes positve 50,50

vMeph, 2017-03-19:
from what i see is that the g54 will folow orders but the UGS X Y Z is not updating the display values in absolute but is displaying in incrementing, bug?

vMeph, 2017-03-21:
@winder  can you clarify on this use i have a lcd that is getting coords from arduino also and the beaivior i see is very diferent
on UGS i see this happening
im at 0,0,0 machine 0,0,0 work
i send X-100  X-100 Z0 and UGS shows -100,-100,0 machine and off course -100,-100,0 work
i save position and off set with G10 P1 L20 X0 Y0 Z0
UGS shows machine 100,100,0 and work 0,0,0
( notice above that the the -100,-100,0 machine become positive 100,100,0  when i hit the G10 P1 L20 X0 Y0 Z0 when it should have been  -100,-100,0 machine
so now if i send lets say 100,100,0 the machine become 200,200,0 and work becomes 100,100,0
when it should be machine 0,0,0 and work 100,100,0
on my lcd is working as expect machine 0,0,0 and work 100,100,0
Ugs is changing from -100 to 100 when you press G10 P1 L20 X0 Y0 Z0
im not able to use the command ? on UGS only say ok, and dont reports nothing but obvious there is a problem on how UGS shows machine coords

winder, 2017-03-21:
The small numbers are the machine coordinates and the large numbers are the work coordinates. Both as reported by GRBL. You can enable verbose output in the preferences to get the detailed console output to see the exact messages coming from grbl.

vMeph, 2017-03-21:
@winder there is a bug on how UGS is showing coords i made some tests and this what im getting

i start machine i send homing $H
[verbose] <Idle|WPos:-5.000,-5.000,-5.000|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000>    UGS shows same

i send command X-100
[verbose] <Idle|WPos:-100.000,-5.000,-5.000|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000> UGS display shows same
 
i set a G54  using folowing command
>>> G10 P1 L20 X0 Y0 Z0
ok
[verbose] <Idle|WPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:-100.000,-5.000,-5.000>  UGS display not shows same now u see machine at X100, Y5,cZ5(it have become positive coords)

now i send $H homing again and
>>> $H
[verbose] <Home|WPos:95.000,0.000,0.000|Bf:15,128|FS:0,0>
ok
[verbose] <Idle|WPos:95.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:-100.000,-5.000,-5.000>
UGS display now shows the folowing machine X195 Y5 Z5 and work coords X95,Y0 Z0

i have reflash  arduino, i have delete temps of UGS platform and the problem is always same, it sounds like UGS platform when issues a G54 makes coords become positive

could you see whats going on or is there something im missing?

winder, 2017-03-22:
It looks like the work position is big and the machine position is small:
```
        this.machinePositionXValue.setFont(small);
        this.machinePositionXValue.setHorizontalAlignment(javax.swing.SwingConstants.RIGHT);
        this.machinePositionYValue.setFont(small);
        this.machinePositionYValue.setHorizontalAlignment(javax.swing.SwingConstants.RIGHT);
        this.machinePositionZValue.setFont(small);
        this.machinePositionZValue.setHorizontalAlignment(javax.swing.SwingConstants.RIGHT);
         
        this.workPositionXValue.setFont(big);
        this.workPositionXValue.setHorizontalAlignment(javax.swing.SwingConstants.RIGHT);
        this.workPositionYValue.setFont(big);
        this.workPositionYValue.setHorizontalAlignment(javax.swing.SwingConstants.RIGHT);
        this.workPositionZValue.setFont(big);
        this.workPositionZValue.setHorizontalAlignment(javax.swing.SwingConstants.RIGHT);
```

I think I see your confusion now when looking at the GRBL status. There is a new format that is a little strange to interpret. To parse this line:
```
[verbose] <Idle|WPos:95.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:-100.000,-5.000,-5.000>
UGS display now shows the folowing machine X195 Y5 Z5 and work coords X95,Y0 Z0
```
The important values are `WPos:95.000,0.000,0.000` (Work Position) and `WCO:-100.000,-5.000,-5.000` (Work Coordinate Offset). Offset of WCO being the important part. Maybe instead of `WCO` it should be called `MCO` (or Just `CO`). In any case, you need to do some arithmatic to get the machine position using `WPos` and `WCO`:
MPosX = 95 - (-100) = 195
MPosY = 0 - (-5) = 5
MPosZ = 0 - (-5) = 5
Or `X195 Y5 Z5`

This is explained here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface

None of this is really useful in helping you with understanding the gcode though. Unfortunately, my machine doesn't having homing switches so I only have basic experience with these commands. I'd recommend this for an overview: https://www.shapeoko.com/wiki/index.php/G-Code#Using_the_Work_Coordinate_Systems

vMeph, 2017-03-22:
but when set work coordinates to 0,0,0  is the small letters that becomes 0,0,0 so will that means that you are setting machine to 0,0,0 in relation to work coords
fells a lilte strange that   wouldnt show has verbose shows and cause alot confusion cause if you going from 0 to -100 and then if you go back to 0 becomes 100 verbose 
seems to work as expected

i guess is just confusing and makes no sense cause in  if im on x-100 and i send it to go  x95 it would pass to the positive space and make double off the travel only strange thing happens to make this a mess is when you hit the command to set G54 it changes from - to +, so if i send to max of my machine dimentions that is 400 when i send back to home again i end up with a machine of 800 dimention hehe

no idea why would it have to be like that when verbose shows how it should be, who knows :(

winder, 2017-03-22:
I just ran the following commands:
```
G0 X-10 Y-10 Z-10
G10 P1 L20 X0 Y0 Z0
```

Afterward, the big number's are 0.00 indicating that the work location has been set to zero, and the small numbers remain at -10. Am I missing something or does this have the opposite behavior for you?

[image]

If I send `G0 X10` after this the large number moves to 10 and the small number moves to 0.

vMeph, 2017-03-22:
to me is doing the oposite behavior, i already reflash arduino and clean start off UGS platform and removed the temps hiden etc.. still happens same no idea :(

chamnit, 2017-03-22:
L20 is set work origin to the XYZ specified. The behavior you're expecting is when you use an L2 instead.

vMeph, 2017-03-22:
@winder ok i have figuer out whats going on

if you set $10=1 you see in UGS the Work coords in big and machine in small and works as expected
if you set $10=2 you see in UGS machine coords in big and work coords in small doesnt work as expected

however in both way verbose works as expected and UGS  display doesnt
dont know but maybe the algorithm that UGS uses is same for both options, dont know if grbl is using diferent algorithms for the response but  it works as expected dont know if is using diferent algorithm

@chamnit  its working now i have used the L20 not sure but from start ouput of grbl verbose was always correct, that was what was making me confused

winder, 2017-03-22:
@vMeph Aah, it looks like I'm calculating MPos wrong when WPos is reported. Notice that it changes between MPos/WPos when you change $10.

closed 2017-03-31 by winder

Ermo10, 2024-04-17:
Hey yall! Quick question. Whenever I run my g-code using UGS, the G54 position is not saved and I have to reset it back to that very position. Right now, I have a line of code that will take my machine back to the G54 position, but I believe there must be an easier way to do this. Thoughts?

commit 717da06f (2017-03-31): Fix Grbl v1.1 parsing with 0=2. Closes #604