winder/Universal-G-Code-Sender #605: Pendant feed rate & UGS jog units

cmkelly01, 2017-03-20:
I'm using the V2.0 nightly (3/13/17), and the pendant defaults to a F1 feed rate, which is sloooow, Is there (or will there be) a way to increase it? Also, will the jog feature for UGS continue to only be in MM, or will the inches option be added back in? You have a great set of software, and a great deal of patience and persistence - my hat's off to you. Thank you so much for keeping at it. I, for one, am very appreciative!!

jahnj0584, 2017-03-20:
Click on the mm box and it'll change the units to imperial

winder, 2017-03-20:
Unfortunately, nobody has really volunteered to help maintain the web pendant UI since it was initially contributed. Web development isn't one of my strengths so I haven't really touched it since the initial contribution.

cmkelly01, 2017-03-20:
Clicking on the mm box just changes it to a "*".

closed 2017-03-20 by cmkelly01

jahnj0584, 2017-03-20:
Want to guess what unit of measurement that signifies?

winder, 2017-03-20:
You see a `*` in the GUI or the Pendant?