winder/Universal-G-Code-Sender #606: Everything comes unresponsive upon using pause or stop buttons

TilmanG, 2017-03-20:
I am able to select a program and run it successfully however when I use the stop button or the pause button the console says "**** Pausing file transfer. ****" or **** Cancelling file transfer. **** and all controls become grayed out. In order to get it "Unstuck" I need to disconnect and reconnect or wait until a dialog box comes up telling me "Grbl timed out" or something similar. 

I'm using a shapeoko 3 with GRBL 0.9. Any help is greatly appreciated!

jahnj0584, 2017-03-20:
Did you try pressing the Play button after?

TilmanG, 2017-03-20:
@jahnj0584 Yes, multiple times however it just keeps repeating the Pausing file transfer message in the console

TilmanG, 2017-03-22:
I fixed the problem by updating the GRBL version on the controller from 0.9 to 1.1

winder, 2017-03-22:
@TilmanG thanks for the update. UGS should still work with 0.9 so I'm going to leave this open for now. I actually noticed this issue before, one of my test Arduino's must be on an older version and I didn't realize it.

winder, 2017-07-09:
0.9 compatibility should be restored now.

closed 2017-07-09 by winder