winder/Universal-G-Code-Sender #607: First movement after changing $100, $101, $102 parameters is always inaccurate

giacomo1574, 2017-03-22:
When I change a step/mm parameter ($100, $101, $102) and I try to move the machine from the keyboard or the GUI controls, the machine moves inaccurately on its first movement. Sometimes exactly twice the supposed distance, sometimes completely randomly. This only happens for the first movement right after I change the value, after that one the axis starts moving perfectly according to the updated value. It's not a big deal but I was wondering if this is a fixable bug of the sender or just a random grbl glitch.

jahnj0584, 2017-03-22:
I only had this happen as the first move is jogged at the old value. I
assumed it's because it needs a "cycle" to apply the setting

winder, 2017-03-22:
@giacomo1574 What version of UGS and GRBL are you using?

Could you attach the command console output when you're setting the command and identify which command isn't working as expected?

closed 2017-03-29 by winder