winder/Universal-G-Code-Sender #610: What did I do wrong ?

labohemia, 2017-03-29:
Hey Guys 

When I open the reader and want to send a Programm or just type $, so just easy commands, it says that GRBL is booting !
ok but GRBL does not boot for several Houers right ? :D 
 And when I am in the serial Monitor from Arduino, it says :

```
 Grbl 0.9a ['$' for help]
ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿerror: Expected command letter
```

pls Help me, I have no idea what to do :/

winder, 2017-03-31:
@labohemia It looks like your EEPROM may have been corrupted which could cause this specific issue. Version 0.9a is extremely old so I would recommend updating to 0.9j or (better) v1.1.

What type of controller board are you using?

closed 2017-04-04 by winder

labohemia, 2017-04-04:
Sorry for not replying I had no Internet Access the last days.
I just downloaded the V.1.1 and it still doesn't work 
I am using the Synthetos GRBL-Shield

thank you for your help !

vMeph, 2017-04-04:
Did you first clean your EEPROM?

labohemia, 2017-04-04:
Yes, I did it once

I have forced the Arduino to reset the trough cabling and delete Everiything and started from the beginning.

I cleaned it for the second time and now it works

I am so happy thank you all :)

but what have could been the problem ?

vMeph, 2017-04-04:
EEprom corrup maybe and when you upgrade from grbl 0.9 to 1.1 you got to clean EEPROM
Glad you fixed your probem!!!