winder/Universal-G-Code-Sender #615: Problem with GUI platform on XP

20051971, 2017-04-06:
Hello,

First all my congratulations for your excellent soft, a big thank to the developers.

I have a small bug  with UGS Platform (Netbeam) that does not work on 3 of my computers and no problem with the other ...
I have the same version of Windows XP, Java on all my computers.

When I run the .exe program nothing happens and any error message???
I can give you the hard configurations of my computers (Chipset, processor, video ...) if it can help you to help me.

The Python version (Classic GUI) work correctly on all my computers

Thanks in advance to all

Frederic

20051971, 2017-04-27:
On 3 different computers on Windows XP (clean installation), the program ugsplatform.exe contained in \ bin does not start without any error message (I do not even have the Netbeans window at startup).

I have done the most tests and I would like to use an Asus EeeBox PC B202:

• CPU: Intel® Atom ™ N270 (CPU speed: 1.60 GHz - Bus speed: 533 MHz - Cache size L2: 512 KB)
• Chipset: Intel® 945GSE ICH7M
• Intel® Graphic Media Accelerator 950 DVMT

I wanted to use this computer a bit old for its small size on my machine CNC 6040 with a Arduino card GRBL 1.1.

I also tried on a Dell SC420 and a Compaq q10 and I have the same problem.

I wonder if this is a Soft or Hard problem because on other computers where I tried it is works correctly without problems (Windows XP, Windows 7 and Windows 10).

I have done some tests, I installed the Kit JDK + NetBeans proposed on the website of Oracle.

The installation goes smoothly and the Netbeans software starts up well, I think this may not be a problem with Netbeans.

Sorry, I don't know the Java programming and Netbeans but I find UGS "Platform" very practical, simple and well done and I would like to be able to use on the mini PC Asus. I find the platform version easier to use than the classic version.

The classic version work correctly no problem with all my computers, I have only the problem with the "Platform" version.

It may just be a mistake in using my part, I apologize.

Thank you in advance for your help

Frederic

winder, 2017-04-27:
@20051971 are there any errors if you run the .exe from your command line?

Are you able to install/run the NetBeans IDE?

It looks like the netbeans platform might not support XP anymore according to this:
https://netbeans.org/community/releases/81/relnotes.html

You could try putting ubuntu on the old netbook, it usually has better support for older hardware.

20051971, 2017-04-28:
Hello,

Thank you for your prompt response.

I will try tonight to run the program on the command line, I just tried from the explorer and start menu. A good idea, there may be a command line error message.

I have installed NetBeans IDE 8.1 as I say above without problem and it run well.

This is curious because on other computers (at least 4 different under Windows XP also) it starts and runs without problems. It can be a problem of incompatibility with the chipset and the graphics card.

I want to keep Windows and do not pass for now on Ubuntu for 2 reasons:
• I do not know at all Unbutu (I know this is not a good argument ....)
• I use specifics software that I do not find on Linux and if I need to make small changes it is simpler for me to do them directly on the small CNC machine rather than go back to the desktop to do them on my main computer.

Some other Windows users like me can already tried and encountered the problem on their machine, have not asked questions and have abandoned this software. In my opinion is a pity because very easy, very simple and I find that  it really deserves to be used by a large number of CNC users.

If everything works well and we find the problem, I could even make a French tutorial for French users to promote your software to French-speaking users.

Thanks again for your help (sorry for my bad English)

Frederic

20051971, 2017-04-28:
Below the error message that I have on the PC that UGS platform does not work:

The launcher has determined that the parent process has a console and will reuse
 it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
Error occurred during initialization of VM
Initial heap size set to a larger value than the maximum heap size

And with the PC of all is OK:

The launcher has determined that the parent process has a console and will reuse
 it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
Loading LocalizingService...
Initializing...
Loading JogService...
Loading ActionService...
Loading MacroService...
Loading StatusLineService...
Services loaded!
Initializing firmware... ...
/resources/firmware_config/
/resources/firmware_config/test_1.json
/resources/firmware_config/smoothieboard.json
/resources/firmware_config/tinyg.json
/resources/firmware_config/grbl.json
/resources/firmware_config/test_2.json
Setting changed
Is enabled: false

winder, 2017-04-28:
@20051971 it looks like the memory settings are the problem. In the `etc` directory, open `ugsplatform.conf` and change the default options:
```
default_options="--branding serverlaunch -J-Xms128m"
```

You can also try lowering 128 to an even smaller number, or adding `-J-Xmx256`.

20051971, 2017-04-28:
In ugsplatform.conf they are not "default_options="--branding serverlaunch -J-Xms128m" but:

default_options="--branding serverlaunch -J-Xms24m -J-Xmx1g
and
default_options="--branding serverlaunch -J-Xms256m

I going to try whit 128M or over value

A big thanks for your help

Frédéric

20051971, 2017-04-28:
Victory it seems to work !!!

Whith "default_options="--branding serverlaunch -J-Xms128m"

It's been 1 month that works on this problem (install on diffrents PC, differents drivers, test versions of Java,...)

I do tomorrow morning more serious tests and keep you informed (it is very cold in my workshop here in France ...)

Thank you verry mutch

Frédéric

20051971, 2017-05-02:
Hello,

Some good news.

This weekend, I have tested the changes of the parameter "default_options =" - branding serverlaunch -J-Xms128m "on all my computers and it works without problems.

I realized several engravings and cuts from complex Gcodes files and everything works fine.

Another thank you and I hope that many users will be able to use this little app very friendly on small configurations.

Frédéric

winder, 2017-09-04:
This should be resolved as of 7140e25829df699e509df7e67f1aea1dd19c7444

closed 2017-09-04 by winder