winder/Universal-G-Code-Sender #617: Issues with return to Zero

jahnj0584, 2017-04-11:
Something weird is happening with my machine. I have 110/110 set to 4200mm/min, and it can cut 5mm pine at that speed without losing steps UNTIL I hit return to zero. The Y axis loses steps and does not operate properly. If I apply the 50% rapid over ride, it DOES work properly. 

I thought it'd be an issue of moving along a hypotenuse (~1.3x regular velocity) but then the issue should also come up while cutting, right?

winder, 2017-04-11:
It sounds like you're losing steps, probably the acceleration settings are different for G1 vs G0. I'd try lowering your acceleration settings until it works at full speed.

jahnj0584, 2017-04-11:
Is there a reason why it'd accelerate differently for G0 though? 

110 = 4200
110 = 4200
120 = 120
121 = 100

winder, 2017-05-10:
If you're still seeing this issue I'd recommend asking in a grbl issue. All I can say for sure is that stepper motors are prone to losing steps when accelerating quickly. Your reasoning sounds good to me, so something must be missing.

closed 2017-05-10 by winder