winder/Universal-G-Code-Sender #618: Stable Release

DirkHeinke, 2017-04-11:
Hello,

do you plan to release a new stable version? The last release is from 2015 and does not work with the new GRBL 1.x
I would love to do some beta testing if you need someone.

jahnj0584, 2017-04-11:
2.0 has been stable and usable for all but one of the versions which was
only out for two days.

winder, 2017-04-11:
@DirkHeinke the Nightly Build is very close to being ready, but I keep getting distracted from the last few items on my checklist. About 40% of UGS downloads are nightly builds, so they are getting a lot of testing. If you'd like to try it out and report any problems you find the download link is at the bottom of this page: http://winder.github.io/ugs_website/download/

I'll try to get a stable release of the 2.0 branch one of these days... I agree, it's been too long.

closed 2017-04-11 by winder

DirkHeinke, 2017-05-02:
There are quite some bugs in our setup but they change every nightly build. Can you perhaps declare a release candidate, so that I/we/all other users can report bugs in relation to that release?

greyltc, 2019-06-04:
Why was this issue closed?
I'm looking for a release here: https://github.com/winder/Universal-G-Code-Sender/releases
Is that the wrong place to look?

JacksCrafts, 2019-06-05:
http://winder.github.io/ugs_website/download/
Try this one.