winder/Universal-G-Code-Sender #621: Add possibility to customizing the toolbar

joacimb, 2017-04-15:
I've been poking around in the code trying to get the toolbar customization working to make UGS more suitable to run on a  computer with a low resolution touch screen. I believe that by making the platform more customizable, UGS can support more platforms and might be easier to extend. 

Take the toolbar panel for handling the GRBL connection for instance. If we break it apart we could hide the settings for firmware or baud-rate as it's something you seldom change. But for some platforms this might be a requirement and would still be supported.

Also, if we try to utilize more functionality from the NetBeans platform we would, as an example, be less dependent on our own action registration procedure and could also get localization and icon-handling by the platform. 

In the picture below i've experimented with the connection and run actions. I've removed the StartStopPauseToolbar and removed the connect-action from the ConnectionToolbar:

[image]

So I would like to try to:

1. Move some of the common actions from ugs-core to ugs-platform/ugs-core and annotate them with NetBeans action annotations. This also requires the constructor that takes the backend parameter to be removed and let them use the CentralLookup instead. The actions also needs to be removed from the RunActionService so that they are not duplicated.
2. Remove ConnectionToolbar and replace it with smaller actions
3. Remove StartStopPauseToolbar and replace it with smaller actions

What do you think?

winder, 2017-04-15:
One of the main reasons so much of the GUI code is in ugs-core is to allow me to support the classic GUI as well as the platform. I would like to use more of the NetBeans platform, and this looks like a big improvement to what I'm currently using, but it would need to be done in a backward compatible way.

About localization, I really didn't like the way it was done with NetBeans - every package needed a separate localization file and couldn't figure out a way to reuse the existing localizations with the platform interface. I'm open to suggestions, as the current approach isn't great (The keyboard shortcut menu cannot be localized). I didn't know about the toolbar customization feature, who knows what other features are broken by the approach I'm using.

Specific to your suggestions:
1. I really like the idea of leveraging more NetBeans platform features, but since I'm still supporting the two interfaces I've been using facades to register standard swing code into the platform rather than migrating it completely. You can see most of the `*TopComponent.java` classes, like `JogControlTopComponent` for how I've been using the NetBeans annotations to register standard swing components.  Doing this with a `StopAction` which wraps `Stop` for instance seems like a great idea.

2. As long as they can still be grouped together by default (I'm assuming the category annotation will handle this), this seems much better than the way I've done it.

3. Sounds good.

winder, 2017-04-15:
I guess the next question is why I'm supporting both GUIs... I think a lot of people find value in the basic no-frills interface of the classic swing implementation. Maybe it will go away eventually but I don't want to deprecate it before even a single stable platform release.

joacimb, 2017-04-16:
I agree that it's important to keep backwards compatibility. I'll give it a go and will try to not break anything. =)

joacimb, 2017-04-17:
The changes are checked in #624, I messed up my commit comments.

closed 2017-05-05 by joacimb